vshymanskyy/TinyGSM #1: why not work on nodemcu with sim800l wst reset after connecting gprs?

Adney, 2016-12-21:
Hi

vshymanskyy, 2016-12-21:
Just committed a fix for that, but i didn't have a chance to check if it works - could you please check if it helps in your situation?

Adney, 2016-12-23:
Thank you.
https://youtu.be/KrreU-3m1-I
Now it is working on mqtt nodemcu with Sim800l.
But in blynk , still not working. Just reboots wst reset nodemcu

Adney, 2016-12-23:
arduino nano with sim800l - don't work blynk.

sketch loaded in the series is empty, nothing happens.

Test Diagnostics is ok, all working.
MQTT is work.

BLYNK is don't working.

vshymanskyy, 2016-12-24:
What do you mean, don't work?

Adney, 2016-12-24:
Nodemcu после загрузки скетча BLYNK, сразу начинает перезагружаться: wst reser.
Arduino NANO после загрузки скетча BLYNK - молчит. В сериал порту пусто, модуль SIM800 не пытается даже установить соединение.

Скетч MQTT клиет - работает.
А вот BLYNK не работает.

closed 2016-12-26 by vshymanskyy

vshymanskyy, 2016-12-26:
Fixed & tested on ESP8266

Adney, 2016-12-27:
Thanks, now BLYNK is work!!!

commit 0d9bae95 (2016-12-21): Fix #1
commit 02213b20 (2016-12-26): Fix #1