vshymanskyy/TinyGSM #2: How can i send a simple SMS?

VittorioEsposito, 2016-12-26:
How can i send a simple SMS? And could you tell me how long does it take to send it?

vshymanskyy, 2016-12-26:
Hi. Currently sending SMS is not supported, but it is planned to be added in future.
PRs are always welcome ;)

vshymanskyy, 2017-01-17:
You can now try:
```  
modem.sendSMS("+380679837464", "some text");
```
or for Unicode (needs data length):
```
  modem.sendSMS_UTF16("+380679837464", u"Привееет!", 10);
```

closed 2017-01-17 by vshymanskyy

facetime88, 2017-06-19:
Hi, how do we get the status result of the sent sms whether success or fail?

roboticboyer, 2017-11-25:
I have tried the function:
`modem.sendSMS("0xxxxxxxxx", "Text here");`
I have received the SMS, but its text is wrong 

The same using:
`modem.sendSMS("0xxxxxxxxx", String ("Text here"));`

Please could you double check?

ClemensGruber, 2018-02-04:
Same problem here with 
`res = modem.sendSMS(SMS_TARGET, "Hello from TinyGSM");`

I got 
`ßeiØΠ-70V`

hubaksis, 2018-02-04:
@ClemensGruber Can you turn on debug and check what is actually going on from the modem's point of view?

ClemensGruber, 2018-02-04:
For 
`  res = modem.sendSMS(SMS_TARGET, "Hello from TinyGSM");
  DBG("SMS:", res ? "OK" : "fail");`

I got on serial 
'SMS: OK'

hubaksis, 2018-02-04:
I mean all AT communication between your controller and the modem.
Check the HttpClient example, it's here:
```
#define DUMP_AT_COMMANDS

...
#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, Serial);
  TinyGsm modem(debugger);
#else
  TinyGsm modem(SerialAT);
#endif

```

Then you'll see all AT communication.

ClemensGruber, 2018-02-04:
An article about SMS encoding: "7 Things You Need to Know about International SMS"
https://www.telesign.com/blog/post/7-things-you-need-to-know-about-international-sms/

I don't konw yet on which position the problem is. I have a German SIM card mobile phone with German UI and also a German operator. Does this matter?

ClemensGruber, 2018-02-04:
I used till now  [AllFunctions.ino](https://github.com/vshymanskyy/TinyGSM/blob/11d068d5297941f1489a2a494f68774057fac6e4/examples/AllFunctions/AllFunctions.ino) as is with all sample / example code. Now I deleted some code that is not necessary before the SMS example and the text is displayed accurate on the phone.

hubaksis, 2018-02-04:
For some reason I can't send SMS at all from my modem. Not sure were issue is. Sorry, can't confirm.
`+CMGS: 14`
I think it indicates the number of characters the modem received.
`Hello from TinyGSM` length is 18. Maybe here is the issue?

Try the following:
(I saw your code before you deleted it)
Change `res = modem.sendSMS("number", "Hello from TinyGSM");` to `res = modem.sendSMS("number", String("Hello from TinyGSM"));`
I'm not a C++ expert, but I think you should send to the function object types that it expects. You are sending char* instead of String.

ClemensGruber, 2018-02-04:
I think I got it, the bad boy is 

```
  String ussd_balance = modem.sendUSSD("*111#");
  DBG("Balance (USSD):", ussd_balance);

  String ussd_phone_num = modem.sendUSSD("*161#");
  DBG("Phone number (USSD):", ussd_phone_num);

```

With this functions the modem is set to [HEX mode](http://m2msupport.net/m2msupport/atcscs-select-te-character-set/) via 
`AT+CSCS="HEX" `

and it is never set back to GSM

hubaksis, 2018-02-05:
Oh, my version already has this fix: 
```
bool sendSMS(const String& number, const String& text) {
    sendAT(GF("+CMGF=1"));
    waitResponse();
    sendAT(GF("+CSCS=\"GSM\""));
    waitResponse();
    sendAT(GF("+CMGS=\""), number, GF("\""));
...
```

ClemensGruber, 2018-02-05:
@hubaksis Thanks for your help! Without your comment and the provoken motivation to have a second look this night I would not have solved this issue -- although it was a stupid cause. ;-)

hubaksis, 2018-02-05:
@ClemensGruber No worries. I'm glad that you found a cause.

ClemensGruber, 2018-02-05:
Seems to be fixed for some modems, e.g. the [M590](https://github.com/vshymanskyy/TinyGSM/pull/42/commits/8431817dde4ca9a489bba80d128bdd44eb0e6bb8#diff-c4684a05e3c97f3da09687ae7b3c85d0)  but not for the SIM800.

hubaksis, 2018-02-05:
I've created a pull request.

ClemensGruber, 2018-02-05:
Spotted already and liked! :-) Thanks!

udayan-santu, 2021-07-06:
### Can anybody pls tell me why this code not sending sms..

`// SIM card PIN (leave empty, if not defined)
const char simPIN[] = "";
// Your phone number to send SMS: + (plus sign) and country code, for Hungary +36, followed by phone number
String mobile_number = "01954004571";
// Configure TinyGSM library
#define TINY_GSM_MODEM_SIM800 // Modem is SIM800
#define TINY_GSM_RX_BUFFER 1024 // Set RX buffer to 1Kb
#include <TinyGsmClient.h>
// TTGO T-Call pins
#define MODEM_RST 5
#define MODEM_PWKEY 4
#define MODEM_POWER_ON 23
#define MODEM_TX 27
#define MODEM_RX 26
// Set serial for debug console (to Serial Monitor, default speed 115200)
#define SerialMon Serial
// Set serial for AT commands (to SIM800 module)
#define SerialAT Serial1
// Define the serial console for debug prints, if needed
//#define DUMP_AT_COMMANDS
#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif
#define IP5306_ADDR 0x75
#define IP5306_REG_SYS_CTL0 0x00

void setup() {
// Set console baud rate
SerialMon.begin(115200);
// Set modem reset, enable, power pins
pinMode(MODEM_PWKEY, OUTPUT);
pinMode(MODEM_RST, OUTPUT);
pinMode(MODEM_POWER_ON, OUTPUT);
digitalWrite(MODEM_PWKEY, LOW);
digitalWrite(MODEM_RST, HIGH);
digitalWrite(MODEM_POWER_ON, HIGH);
// Set GSM module baud rate and UART pins
SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
delay(3000);
// Restart SIM800 module, it takes quite some time
// To skip it, call init() instead of restart()
SerialMon.println("Initializing modem");
modem.init();

}
void loop() {
modem.sendSMS("+8801954004571", "some text");
delay(10000);
}`

SRGDamia1, 2021-07-06:
You probably need to check your network registration status first.

commit 39b563fc (2017-01-17): Fix #3, Fix #2