vshymanskyy/TinyGSM #3: Any chance you would be able to add AT+CIPGSMLOC ?

evmond, 2016-12-28:
Hi There, 

Big fan of your work here. Was wondering if there is any chance of adding AT+CIPGSMLOC for location triangulation. Would be great for IoT dashboards which have map widgets

VittorioEsposito, 2016-12-28:
@evmond Working on it in my library

VittorioEsposito, 2016-12-28:
@evmond Just added [here](https://github.com/VittorioEsposito/Sim800L-Arduino-Library-revised/releases)

vshymanskyy, 2016-12-29:
Sure, will be added. This stuff is much simple than providing proper GPRS support ;)

ethernutza, 2017-01-04:
Hello

For a noob like me this is a winner! Thank you very much! Any chance you can add getIMEI to this library or is that pushing it?

vshymanskyy, 2017-01-17:
Added some functions:
```
  modem.setGsmBusy();
  String imei = modem.getIMEI();
  int signal = modem.getSignalQuality();
  String loc = modem.getGsmLocation();
```

closed 2017-01-17 by vshymanskyy

ethernutza, 2017-01-17:
Brilliant! Thank you very much!

sanjulanipun, 2023-02-23:
hi @vshymanskyy  any possible to add `AT+CNUM` for sim7600 to get sim card number ?

or any way have to send that custom at command with this library and get a response and filter it?

vshymanskyy, 2023-02-23:
Yes, both are possible ;) will look into it

commit 39b563fc (2017-01-17): Fix #3, Fix #2