vshymanskyy/TinyGSM #4: Freeze after about 15 Minutes

GeorgianBorca, 2017-01-05:
After about 10-15 minutes, with a few reconnectings in between my code stops working. The ardino dose not receive anything and dose not transmit anything anymore. I have to manualy reset it to get it working again.  
Dose anybody managed to get anything stable with this library?
I'm using sim800H with arduino running on 8Mhz. The GSM modem works just fine and i don't think is anything related to my code. 
The brocker i use is : broker.hivemq.com
Can it be something related to the brocker?
Thanks.

vshymanskyy, 2017-03-27:
@GeorgianBorca , plese tell me if you have overcome the issue, or investigated it further.

GeorgianBorca, 2017-03-30:
The maximum time i got is 3 days. I will eneable the debug function and post the results.

vshymanskyy, 2017-09-02:
Please reopen if this still happens on master branch. Thank you!

closed 2017-09-02 by vshymanskyy