vshymanskyy/TinyGSM #7: A6

spa-sam, 2017-01-17:
Very cool! I hope that support for the A6 too? Thank you for the work done!

vshymanskyy, 2017-01-18:
Yes, especially if someone helps porting it ;) Shouldn't be hard but requires 4-5 hours of tinkering.
I currently don't have neither time not the hardware to start with :(

spa-sam, 2017-01-18:
Unfortunately not much programming experience, just learning. I can send unit in Ukraine.
К сожалению не так много опыта в программировании, только учусь. Могу отправить модуль, по Украине.

vshymanskyy, 2017-01-19:
No thanks, I'll get mine soon. Time is much more critical ;)

vshymanskyy, 2017-02-01:
Basic A6 support is implemented.
More testing is needed, and community support in bugfixing, improving stability is very welcome... ;)

closed 2017-02-14 by vshymanskyy