vshymanskyy/TinyGSM #9: knowing Local IP Address

mushfiq-cse05-buet, 2017-01-30:
is there any function in TinyGsm class so that i can get the Local IP address assigned by GSM operator after  calling gprsConnect() function ????? 
Thank you!

vshymanskyy, 2017-02-01:
Which modem model do you need it for?

mushfiq-cse05-buet, 2017-02-04:
SIM800L please :) @vshymanskyy

mushfiq-cse05-buet, 2017-02-04:
i think you can add up functions like getLocalIP for eery model :) thanks @vshymanskyy

bsiege, 2017-04-23:
SIM800//SIM900
Serial3 << (F("AT+CIFSR")) << endl;//get local IP adress
But mostly this will not help because it is CGNAT'ted to public internet.
For public IP you need some external service, which could not longterm reliable and data consuming. In my case i prefer to log sensor IP on the receiver (server) side.

commit 4a86db07 (2017-09-10): Fix gprsConnect, add getLocalIP(). Fix #9
closed 2017-09-10 by vshymanskyy