vshymanskyy/TinyGSM #11: SIM5320

MattLeMasureir, 2017-02-07:
Thank you for your work on this so far. Can't wait to get my projects working over cellular networks!

If you had time to implement the SIM5320 I know a lot of people would appreciate it - Adafruit haven't yet completed a library for this module in their FONA 3G break-out.

vshymanskyy, 2017-02-14:
I need someone to donate a SIM5320 device to work on that :)

GeorgianBorca, 2017-02-14:
I could donate you one but you'll need to make a PCB for it :) I also have an SIM7100E (LTE) module. Let me know if i can help.

vshymanskyy, 2017-02-14:
unfortunately no time for PCB etching ;)

GeorgianBorca, 2017-02-16:
The SIM5320 hase an MQTT built in so there is no need for this library. Dose anybody have a module to test the AT commands? 
http://www.microchip.ua/simcom/WCDMA/APPNOTES/SIMCom_3G_MQTT_Application%20Note_V1.00.pdf

FabianInostroza, 2017-05-19:
@GeorgianBorca I have a lot of SIM5320A modules with firmware version 1575B13SIM5320A and they don't support those commands. According to one of the authorized sellers that is the latest firmware version.

GeorgianBorca, 2017-05-24:
@FabianInostroza Thank you. Mabe there is some other version of the firmware with MQTT in it. Like with SIM800H some firmware has bluetooth support some don't.

vshymanskyy, 2017-09-02:
Closed due to lack of interest

closed 2017-09-02 by vshymanskyy

dashracer, 2017-10-15:
Really need this @vshymanskyy ,

If I want to add feature for sim5320E, could you please give me clue what should I modify?
I need this to work with MQTT PubSubClient

kamrulroky, 2017-11-01:
This would be very helpful if this library includes SIM5320E as a supported device , with sim800l mqtt works fine but there is some reconnecting issue . A 3G module would work better I think.

SRGDamia1, 2017-11-03:
Just FYI, Digi International has a 3G (3G HSPA/GSM with 2G fall-back) XBee that's carrier certified for AT&T in the US:  https://www.digi.com/products/xbee-rf-solutions/embedded-cellular-modems/digi-xbee-cellular-3g.  It retails for $89, which is only $10 more than the Adafruit Fona 3G.  It should work just find with the "XBee" module that already is in this library.  The cellular chip inside is a u-blox 201, so if you would rather use "bypass mode" and interact directly with the cellular chip instead of using the XBee command interface, you also have that option.

I haven't tried MQTT on my board, but if you can use TinyGSM with a SIM800 to get to MQTT, you should also be able to use the 3G XBee for it.

SRGDamia1, 2017-11-03:
As soon as they are available, I'm hoping to grab some of the Digi LTE CAT-M XBee's.  I also picked up a NimbeLink LTE module based on the Telit LE910, but I have no idea when (if?) I'll ever have a chance to get it working for this library.

jongips, 2017-12-13:
I'm also interested in support for the NimbeLink / Telit LTE 910 series modems. @vshymanskyy Any thoughts on how much effort it would be to port support? I'm considering diving in.

SRGDamia1, 2017-12-13:
I haven't had any time to do anything with the NimbeLink / Telit LTE 910; it's in a box on my office floor.  Porting over most of the functions is just a matter of selecting the equivalent AT commands.  A lot of the commands to connect to the network and open sockets are fairly similar between modules.  What I found to be the trickiest was figuring out how the module tells you when data/response has been received over a socket and transferring that data over to the Arduino "client."  You have to re-write the "wait response" and read commands to deal with the format of the response and (for me at least) that took a bit of fiddling and testing.

hubaksis, 2018-02-16:
@SRGDamia1 Can you please share your code for SIM5320?
I am using CHTTPS to send data to server, however, I am struggling with closing connection after a request. 
`AT+CHTTPSCLSE ` and `AT+CHTTPSSTOP ` always return error unless the server closes the connection first. When the server closes connection CHTTPSSTOP returns OK and I can make another request.

SRGDamia1, 2018-02-17:
I'm sorry, I don't have a SIM5320 or any code for it.  The 3G module I've tested is the Digi 3G XBee.

Najab, 2019-03-06:
bool gprsConnect(const char* apn, const char* user = NULL, const char* pwd = NULL) {
    gprsDisconnect();

    // Set the Bearer for the IP
    sendAT(GF("+SAPBR=3,1,\"Contype\",\"GPRS\""));  // Set the connection type to GPRS
    waitResponse();

    sendAT(GF("+SAPBR=3,1,\"APN\",\""), apn, '"');  // Set the APN
    waitResponse();

    if (user && strlen(user) > 0) {
      sendAT(GF("+SAPBR=3,1,\"USER\",\""), user, '"');  // Set the user name
      waitResponse();
    }
    if (pwd && strlen(pwd) > 0) {
      sendAT(GF("+SAPBR=3,1,\"PWD\",\""), pwd, '"');  // Set the password
      waitResponse();
    }

    // Define the PDP context
    sendAT(GF("+CGDCONT=1,\"IP\",\""), apn, '"');
    waitResponse();

    // Activate the PDP context
    sendAT(GF("+CGACT=1,1"));
    waitResponse(60000L);

    // Open the definied GPRS bearer context
    sendAT(GF("+SAPBR=1,1"));
    waitResponse(85000L);
    // Query the GPRS bearer context status
    sendAT(GF("+SAPBR=2,1"));
    if (waitResponse(30000L) != 1)
      return false;

    // Attach to GPRS
    sendAT(GF("+CGATT=1"));
    if (waitResponse(60000L) != 1)
      return false;

    // TODO: wait AT+CGATT?

    // Set to multi-IP
    sendAT(GF("+CIPMUX=1"));
    if (waitResponse() != 1) {
      return false;
    }

    // Put in "quick send" mode (thus no extra "Send OK")
    sendAT(GF("+CIPQSEND=1"));
    if (waitResponse() != 1) {
      return false;
    }

    // Set to get data manually
    sendAT(GF("+CIPRXGET=1"));
    if (waitResponse() != 1) {
      return false;
    }

    // Start Task and Set APN, USER NAME, PASSWORD
    sendAT(GF("+CSTT=\""), apn, GF("\",\""), user, GF("\",\""), pwd, GF("\""));
    if (waitResponse(60000L) != 1) {
      return false;
    }

    // Bring Up Wireless Connection with GPRS or CSD
    sendAT(GF("+CIICR"));
    if (waitResponse(60000L) != 1) {
      return false;
    }

    // Get Local IP Address, only assigned after connection
    sendAT(GF("+CIFSR;E0"));
    if (waitResponse(10000L) != 1) {
      return false;
    }

    // Configure Domain Name Server (DNS)
    sendAT(GF("+CDNSCFG=\"8.8.8.8\",\"8.8.4.4\""));
    if (waitResponse() != 1) {
      return false;
    }

    return true;
  }

I just want the replace this function for SIM5320 but i tried something please check and correct it now my function is not working 

bool gprsConnect(const char* apn, const char* user = NULL, const char* pwd = NULL) {
    //gprsDisconnect();

    //// Set the Bearer for the IP
    //sendAT(GF("+SAPBR=3,1,\"Contype\",\"GPRS\""));  // Set the connection type to GPRS
    //waitResponse();

    //sendAT(GF("+SAPBR=3,1,\"APN\",\""), apn, '"');  // Set the APN
    //waitResponse();

    //if (user && strlen(user) > 0) {
    //  sendAT(GF("+SAPBR=3,1,\"USER\",\""), user, '"');  // Set the user name
    //  waitResponse();
    //}
    //if (pwd && strlen(pwd) > 0) {
    //  sendAT(GF("+SAPBR=3,1,\"PWD\",\""), pwd, '"');  // Set the password
    //  waitResponse();
    //}

	sendAT(GF("+CGATT=1"));  // Set the connection type to GPRS
	waitResponse();

	sendAT(GF("+CGACT=1,1"));  // Set the connection type to GPRS
	waitResponse();

	//sendAT(GF("+CGATT=1"));  // Set the connection type to GPRS
	//waitResponse();

    // Define the PDP context
	sendAT(GF("+CGDCONT=1,\"IP\",\""), apn, GF("\",\"0.0.0.0\""));
    //sendAT(GF("+CGDCONT=1,\"IP\",\""+String(apn)+"\",\"0.0.0.0\""));
    waitResponse();

    // Activate the PDP context
    sendAT(GF("+CGACT=1,1"));
    waitResponse(60000L);

    //// Open the definied GPRS bearer context
    //sendAT(GF("+SAPBR=1,1"));
    //waitResponse(85000L);
    //// Query the GPRS bearer context status
    //sendAT(GF("+SAPBR=2,1"));
    //if (waitResponse(30000L) != 1)
    //  return false;

	sendAT(GF("+CGPADDR=1"));
	if (waitResponse(30000L) != 1)
		return false;

    // Attach to GPRS
    sendAT(GF("+CGATT=1"));
    if (waitResponse(60000L) != 1)
      return false;

    // TODO: wait AT+CGATT?

    //// Set to multi-IP
    //sendAT(GF("+CIPMUX=1"));
    //if (waitResponse() != 1) {
    //  return false;
    //}

    //// Put in "quick send" mode (thus no extra "Send OK")
    //sendAT(GF("+CIPQSEND=1"));
    //if (waitResponse() != 1) {
    //  return false;
    //}

    //// Set to get data manually
    //sendAT(GF("+CIPRXGET=1"));
    //if (waitResponse() != 1) {
    //  return false;
    //}

    //// Start Task and Set APN, USER NAME, PASSWORD
    //sendAT(GF("+CSTT=\""), apn, GF("\",\""), user, GF("\",\""), pwd, GF("\""));
    //if (waitResponse(60000L) != 1) {
    //  return false;
    //}

    //// Bring Up Wireless Connection with GPRS or CSD
    //sendAT(GF("+CIICR"));
    //if (waitResponse(60000L) != 1) {
    //  return false;
    //}

    //// Get Local IP Address, only assigned after connection
    //sendAT(GF("+CIFSR;E0"));
    //if (waitResponse(10000L) != 1) {
    //  return false;
    //}

    //// Configure Domain Name Server (DNS)
    //sendAT(GF("+CDNSCFG=\"8.8.8.8\",\"8.8.4.4\""));
    //if (waitResponse() != 1) {
    //  return false;
    //}

    return true;
  }