vshymanskyy/TinyGSM #12: GPRS Status

alvarolb, 2017-02-08:
Hi! First of all, congratulations for your library! It just works! I am integrating it with the thinger.io platform, and got nice results! :)

Just one question. My device must be connected indefinitely to a socket, and the device have some mechanisms to detecting connection loss, autoreconnect, and so on. I have managed to check the current network status by calling the getRegistrationStatus(), that let me avoid registering again to the network if the modem is already connected to it. I need a similar check for the GPRS Apn connection. The socket may be closed by the remote endpoint, but my modem can still be connected to the APN. Do you know if there is any way to check if the modem is connected to the APN to avoid reconnecting again? I am trying to optimize and minimize the network operations...

Thanks!

vshymanskyy, 2017-02-14:
Great. I think it can be done, but the AT commands may be not that simple. Sometimes it's just easier to reconfigure everything. Still, I'm open for discussions and supporting contributions in this area.

vshymanskyy, 2017-03-27:
Did you get it working? Is there room for improvement?

bsiege, 2017-04-23:
i am investigating for a GSM-Library to replace my working but ugly grown code. This one seems very promising. The problem from alvarolb is well known. I could solve this via CIPACK for sending. If transmission is broken, ACK's are missing and i can decide to try a reconnect. 
Manual: (SIM900, SIM800)
```
Response
+CIPACK: <txlen>, <acklen>, <nacklen> OK
Parameters
  
<n> A numeric parameter which indicates the connection number 
<txlen>The data amount which has been sent
<acklen>The data amount confirmed successfully by the server
<nacklen> The data amount without confirmation by the server
```
This parameter is very helpful for connection watchdog. Especially when firing data to a socket and nacklen is growing. But i have not (yet) enough knowledge how to integrate this into a library.

```
void ConnWatchDog () {
  if (elapLastConnWatchDog >= ConnWatchDogIV && elapSinceON >= 15000 && powerSaveMode == false) {
    ClearSerialData(); //Buffer Leermurksen

    //Check transferred packets
    Serial3 << (F("AT+CIPACK")) << endl;
    ABdelay(500);

    Serial3.findUntil("+CIPACK", "\n\r");
    cipack1 = Serial3.parseInt(); //txlen
    cipack2 = Serial3.parseInt(); //acklen
    cipack3 = Serial3.parseInt(); //nacklen

    if (cipack1 > lastcipack1 ) {
      isSending = 1;
      softresets = 0;
    }
```

mrWheel, 2017-05-10:
@vshymanskyy
great library!!

Is there a way to set a timeout for waitForNetwork() and gprsConnect()?

closed 2017-09-28 by vshymanskyy