vshymanskyy/TinyGSM #13: A6 don't connect APN

adz1122, 2017-02-09:
I test found A6 don't connect APN,always fail, I used MQTT example, but AT_DBUG is OK

vshymanskyy, 2017-02-14:
Please provide steps to reproduce, what output you see and what is expected.

widoyo, 2017-02-17:
I use Diagnostic sketch. Serial monitor stop after "Waiting for Network". I use A6 Module modem with BAUDrate 115200.

```
+CPIN:READ\

OK

+CTZV917/02/17,22911:51,+07

+CIEV: sersice,  1
+CL�V: roam, 0
+CREG: 1Waiting for network...AT+CREG?

+CREGz 1,1

OKAT+CREG?

+CREG9 1,1

OK
	AT+CREG?

�(�Q�'�b�j
OK
AT+CREG?

+CREG9 1,1

OK
	AT+CREG?

+A�EG: 1,1

	OK
AT+CREG?

+A�EG: 1,1

	OK
AT+CREG?

+(�Q�'�b�j

	OK
AT+CREG?

+A�EG: 1,1

	OK
AT+CREG?

+A�EG: 1,1
OK
AT+CREG?

�(�Q�'�b�j
OK
AT+CREG?

+A�EG: 1,1

OK
AT+CREG?

+CREG: 1.1

OK

```

GeorgianBorca, 2017-02-17:
To me, it looks like you have a lot of noise in there. Try a lower boudrate and shorter cables. Avoid power cables near the data cables.

widoyo, 2017-02-18:
I use Arduino UNO (5v) without voltage divider on RX of A6. May be thats
the problem.

widoyo, 2017-02-18:
I set lower **BAUDRATE into 9600** and better result.

closed 2017-03-07 by vshymanskyy