vshymanskyy/TinyGSM #14: A6/A7 call status [not a issue]

reaper7, 2017-02-17:
Hi,
i have a question about a6/a7 future implementation,
You are planning to add call status information?
Something like in this lib:
https://github.com/skorokithakis/A6lib/blob/master/A6lib.h#L30

I try to build car location logger with some alarm functions
and Your lib looks promising, at this moment I can upload 
data to remote server (I hope that getLocation will be added soon)

But very desirable fuction is a callstatus, when A7 module call number
then I need feedback inside code if recipient picked up or reject or not answer etc...

What about receive sms? It would be also a good option for sending commands to module.

regards!

vshymanskyy, 2017-02-20:
Recieve SMS is not currently planned.
getLocation - it is only available on some modules... do you mean GSM or GPS/GNSS location?
Other functions should be added sooner or later.

reaper7, 2017-02-20:
I mean GPS

vshymanskyy, 2017-02-21:
GPS support is not currently planned, but Pull Requests are more than welcome. Thx

vshymanskyy, 2017-02-22:
BTW, @reaper7 does A6/A7 GPRS work for you?

reaper7, 2017-02-22:
I do not have A7 module yet. I waiting for elecrow atmel 32U4 with A7 GPS+GSM/GPRS
I wrote that "I can upload data" in meaning at current state of libraries.

vshymanskyy, 2017-03-27:
Any attempts to port some functionality from A6lib are welcome - please send PRs.
I'll help reviewing/testing/creating examples.

reaper7, 2017-03-28:
I received my module (32U4 with A7) last week, I do not have much time but I've already done some tests.
For now only related with http get requests. Looks like A7 is slower than SIM800.
There aren't also some useful commands AT+HTTPxxxxx

vshymanskyy, 2017-03-28:
@reaper7 - TinyGSM examples should work with A6 and A7 - please tell me if it works for you!

reaper7, 2017-03-28:
Yes, for e.g WebClient works ok for me.
Of course with little addition, because this module need to set some 32u4 pins for A7 control. (pwr/rst/sleep)

achillesh, 2017-04-12:
Can you give me an example of use of pwr/rst/sleep on a7 with 32U4 by TinyGSM ?

reaper7, 2017-04-12:
@achillesh - I do not have example for this,
only simply initialization suitable pins inside setup, that's all.
Look at example on elecrow wiki: https://www.elecrow.com/wiki/index.php?title=32u4_with_A7_GPRS/GSM

sometging like this:

```
void setup()
{
  Serial.begin(115200);
  Serial1.begin(115200);
  pinMode(4, OUTPUT); //RESET
  pinMode(5, OUTPUT); //SLEEP
  pinMode(8, OUTPUT);  //PWR_KEY
  digitalWrite(5, HIGH); 
  digitalWrite(4, LOW); 
  digitalWrite(8, LOW); 
  Serial.println("After 2s, power on process!!");
  delay(2000);
  Serial.println("power on begin!!");
  digitalWrite(8, HIGH); 
  delay(3000);       
  digitalWrite(8, LOW);
  Serial.println("A7 Power ON!");  
}

void loop() {

}
```

achillesh, 2017-04-12:
I'm working with the above start up and then it hangs 

After 2s, power on process!!
power on begin!!
A7 Power ON!
Initializing modem...
Waiting for network...

any idea?

reaper7, 2017-04-12:
I can't test this now... i'm outside home...

achillesh, 2017-04-12:
Thanks a lot..

reaper7, 2017-04-12:
so...webclient works for me:

this is reply:
```
After 2s, power on process!!
power on begin!!
A7 Power ON!
Initializing modem...
Waiting for network... OK
Connecting to internet OK
Connecting to wtfismyip.com OK
HTTP/1.1 200 OK
X-Hire-Me: clint@wtfismyip.com
Content-Type: text/plain; charset=utf-8
Cache-Control: no-cache, no-store, max-age=0, must-revalidate
Pragma: no-cache
Expires: 0
Content-Length: 14
Date: Wed, 12 Apr 2017 20:42:15 GMT
Connection: clo
Server disconnected
GPRS disconnected
```

this is code (replaced APN and server):
```
/**************************************************************
 *
 * This sketch connects to a website and downloads a page.
 * It can be used to perform HTTP/RESTful API calls.
 *
 * TinyGSM Getting Started guide:
 *   http://tiny.cc/tiny-gsm-readme
 *
 **************************************************************/

// Select your modem:
//#define TINY_GSM_MODEM_SIM800
//#define TINY_GSM_MODEM_SIM900
#define TINY_GSM_MODEM_A6
//#define TINY_GSM_MODEM_M590

#include <TinyGsmClient.h>

// Your GPRS credentials
// Leave empty, if missing user or pass
const char apn[]  = "internet";
const char user[] = "";
const char pass[] = "";

// Use Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial1

// or Software Serial on Uno, Nano
//#include <SoftwareSerial.h>
//SoftwareSerial SerialAT(2, 3); // RX, TX

TinyGsm modem(SerialAT);
TinyGsmClient client(modem);

//const char server[] = "cdn.rawgit.com";
//const char resource[] = "/vshymanskyy/tinygsm/master/extras/logo.txt";

const char server[] = "wtfismyip.com";
const char resource[] = "/text";

int port = 80;

void setup() {
  // Set console baud rate
  Serial.begin(115200);
  delay(10);

  // Set GSM module baud rate
  SerialAT.begin(115200);
  delay(3000);

  //a7 setup
  pinMode(4, OUTPUT); //RESET
  pinMode(5, OUTPUT); //SLEEP
  pinMode(8, OUTPUT);  //PWR_KEY
  digitalWrite(5, HIGH); 
  digitalWrite(4, LOW); 
  digitalWrite(8, LOW); 
  Serial.println("After 2s, power on process!!");
  delay(2000);
  Serial.println("power on begin!!");
  digitalWrite(8, HIGH); 
  delay(3000);       
  digitalWrite(8, LOW);
  Serial.println("A7 Power ON!");
  //a7 setup done 

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  Serial.println(F("Initializing modem..."));
  modem.restart();

  // Unlock your SIM card with a PIN
  //modem.simUnlock("1234");
}

void loop() {
  Serial.print(F("Waiting for network..."));
  if (!modem.waitForNetwork()) {
    Serial.println(" fail");
    delay(10000);
    return;
  }
  Serial.println(" OK");

  Serial.print(F("Connecting to "));
  Serial.print(apn);
  if (!modem.gprsConnect(apn, user, pass)) {
    Serial.println(" fail");
    delay(10000);
    return;
  }
  Serial.println(" OK");

  Serial.print(F("Connecting to "));
  Serial.print(server);
  if (!client.connect(server, port)) {
    Serial.println(" fail");
    delay(10000);
    return;
  }
  Serial.println(" OK");

  // Make a HTTP GET request:
  client.print(String("GET ") + resource + " HTTP/1.0\r\n");
  client.print(String("Host: ") + server + "\r\n");
  client.print("Connection: close\r\n\r\n");

  unsigned long timeout = millis();
  while (client.connected() && millis() - timeout < 15000L) {
    // Print available data
    while (client.available()) {
      char c = client.read();
      Serial.print(c);
      timeout = millis();
    }
  }
  Serial.println();

  client.stop();
  Serial.println("Server disconnected");

  modem.gprsDisconnect();
  Serial.println("GPRS disconnected");

  // Do nothing forevermore
  while (true) {
    delay(1000);
  }
}
```
btw ... I don't know why I only see HEADER reply but not CONTENT (IP address as plain text in this example)

achillesh, 2017-04-29:
I want your opinion about A7 32u4 (i got the same board with yours). I 'm working to set it in low power mode but it never goes below 9.5ma.
There is something that i don' understand with A7.. a positive pulse on PWRKEY (3sec) means power on and then it is open. We must use the same procedure to set it power off (PWRKEY 3sec pulse) or we can simply send AT+COPF. Even tought the board remains on 100ma after power on procedure. Any idea 
P.S.: i made some simple addon functions to your great lib (time & location at the moment)

abhishekshil, 2017-06-23:
can some please provide me with a program for getting complete coordinate of the GPS provided in the 32u4 with A7 module ASAP
Thanks in advance

abhishekshil, 2017-06-23:
And if tinyGPS works then wht are the modification to the pin should be done to make it work

kooliha, 2017-08-06:
Hi Achillesh

Did you sort out how to work with low power mode? For me, it does not work.
Please help

Thanks

achillesh, 2017-09-21:
The lowest power for me is 0.65mA..

vshymanskyy, 2017-09-28:
Closing this for now.. I have implemented an ability to perform outgoing calls on A6/A7 & SIM800 series.
Incoming notification is planned, but no estimates for now.

closed 2017-09-28 by vshymanskyy