vshymanskyy/TinyGSM #17: Board compatibility

flowmeter, 2017-03-08:
Hi
Have someone checked compatibility with this board [http://www.tinyosshop.com/index.php?route=product/product&product_id=464](url) ?
I tried serial connection check and diagnostic sketch, none worked.
I can use my board with the library provided by Tinysine, and can send SMS. Led status tells mobile network is OK.
Yet original library has not enough functions to POST on [data.sparkfun.com](url) using phant.io library.

vshymanskyy, 2017-03-13:
This is SIM900, so should work after you setup Arduino <> module communication properly.

flowmeter, 2017-03-13:
Thank you for your reply,
They are plugged together and straps are set to software serial communication. Is there anything else to do?

vshymanskyy, 2017-03-13:
What does diagnostic sketch output?
Are you able to get AT commands from arduino to the device at all?

vshymanskyy, 2017-03-27:
Closed due to lack of interest

closed 2017-03-27 by vshymanskyy