vshymanskyy/TinyGSM #18: Adafruit Fona 808 (SIM808) on Arduino mega

larsrengersen, 2017-03-09:
I have an Arduino Mega 2560 and a Adafruit Fona 808 shield.
When I use the Adafruit test sketch the serial monitor says "FONA is OK" and "Found FONA 808 (v2)" and I can run the test cases like read the sim ID, etc.
So I assume the wiring, available power etc all is OK.
Even though I am using the Mega I am using SoftwareSerial since Adafruit Fona currently does not support hardware serial. Next to the pins that work for the Adafruit test sketch, I have tried all Mega serial pins without any luck.
While using AT_debug all baud rates fail.
Do you have a suggestion what to do/check next?

vshymanskyy, 2017-03-13:
AT_debug all baud rates fail - this means you need to figure out the right SW/HW configuration.
Sorry I don't own an Adafruit Fona 808 (SIM808), so can't test it.
Maybe someone from the community can help.

GeorgianBorca, 2017-03-13:
Quote from arduino.cc website: 

> Not all pins on the Mega and Mega 2560 support change interrupts, so only the following can be used for RX: 10, 11, 12, 13, 14, 15, 50, 51, 52, 53, A8 (62), A9 (63), A10 (64), A11 (65), A12 (66), A13 (67), A14 (68), A15 (69).  <

Are you using the right pins?

larsrengersen, 2017-03-13:
@vshymanskyy : Thanks for the hint, I will dive into SW/HW (lots to learn ;) )
@GeorgianBorca : Thanks, yes I am using the right pins.
With Adafruit library the following work
`#define FONA_RX 50
#define FONA_TX 51
#define FONA_RST 22`
But I've tried others as well without any success (so far).

larsrengersen, 2017-03-13:
Using the Adafruit library and example I can get AT commands from the Mega to the Fona.
After booting up it says Call Ready and SMS Ready.
"pressing" G via the Serial Monitor means 'Enable GPRS' and works.
This basically is the following code from the example:
`   case 'G': {
        // turn GPRS on
        if (!fona.enableGPRS(true))
          Serial.println(F("Failed to turn on"));
        break;
      }`
> FONA> 
Call Ready

SMS Ready
G
	---> AT+CIPSHUT
	<--- SHUT OK
	---> AT+CGATT=1
	<--- OK
	---> AT+SAPBR=3,1,"CONTYPE","GPRS"
	<--- OK
	---> AT+SAPBR=3,1,"APN","FONAnet"
	<--- OK
	---> AT+CSTT="FONAnet"
	<--- OK
	---> AT+SAPBR=1,1
	<--- OK
	---> AT+CIICR
	<--- OK

larsrengersen, 2017-03-13:
@vshymanskyy 
Since it does work with the Adafruit library I tend to think that the TinyGSM library isn't compatible, can that be the case?
How do I further check on SW/HW configuration? Google isn't helpful.

> Sorry I don't own an Adafruit Fona 808 (SIM808), so can't test it.
I can send you mine by post if that would help and you are willing to further look into it.

GeorgianBorca, 2017-03-14:
Unlike the fona library, this library dose not support hardware reset and hardware power ON/OFF.  The KEY pin should be connected to GND so the board will power ON straight away. I advise you to add some 5 seconds delay at the begining of void setup().  If you can share your code i could test it using one SIM808 board (not a fona).

larsrengersen, 2017-03-14:
Thanks @GeorgianBorca that was a promising reply.
However, I found out that I already grounded key.
The delay(5000); is also already in place.
My code currently just is the [AT_debug tool from the TinyGSM library](https://github.com/vshymanskyy/TinyGSM/tree/master/tools/AT_Debug) with hardware serial commented out and software serial enabled with correct pin numbers entered.
I double checked my solderings but unfortunately those are all OK too.
The net led on the module blinks 64 ms on and 3 seconds off, which means "the module has made contact with the cellular network and can send/receive voice and SMS".
But it does not yet "64ms on, 300ms off - the GPRS data connection you requested is active".
But I'm not sure whether that is needed for the baud rate test already.

GeorgianBorca, 2017-03-14:
Intersting. I will make a test skecth using this library with my boards and post it so you can test it. 
I remember having having some issues with software serial and a GSM modem. I could send data to it but everythign i received was just rubish characters. After hours and hours i found that the module was using a wrong baudrate by default. However i burnt a new firmware on it with autbaud as defauld and it worked just fine. 
Meanwhile could you connect the board staight to the PC? or find another way to issue AT command and get a **very long response.** Issuing **AT&F** command you will get a lot of data. Those are the default ettings. Look for IPR+ line and see if it's 0 (zero). if it's not 0 you don't have the autobaud enabled. You could send AT+IPR=0 and than AT&W to store this setting. 
Hope this helps. 

On arduino mega you could connect the board to Serial1 and in loop have the code:
`if (Serial.available() > 0) {
Serial1.write(Serial.read());
}
if (Serial1.available() >0) {
Serial.write(Serial1.read());
}`
Use 115200 for Serial and 9600 for Serial1.

larsrengersen, 2017-03-15:
Thanks, I have ordered a USB TTL cable to be able to hook the module up directly to my computer and be able to send direct AT commands. Cable should arrive in a couple of days.
The Fona module should be auto-baud but you never know.

larsrengersen, 2017-03-20:
First of all, I also ordered a Fona 808 breakout and tried that one. 
Unfortunately no success using AT_debug. Still all baud rates fail, like with the shield.

Furthermore, I have received the USB cable and was able to successfully connect to the Fona 808.
On the command "AT&F", it just said OK.
What parameter do I need to add to get the long response including IPR+ value?

GeorgianBorca, 2017-03-20:
The command was  AT&V sorry. the AT&F command restores the factory settings. see page 47 on this document.
https://www.elecrow.com/download/SIM800%20Series_AT%20Command%20Manual_V1.09.pdf

larsrengersen, 2017-03-20:
Thanks @GeorgianBorca AT&V did do the trick and gave me the profile.
"Unfortunately" +IPR : 0 was already set.
Any suggestions what I can try/do next?

GeorgianBorca, 2017-03-27:
I don'nt see any problem if you got that anser over serial. Just use that baudrate in MQTT test sketch of webclient and you are good to go.

larsrengersen, 2017-03-27:
Well my problem is that I do get an answer when I use the Adafruit Fona library using Fona+Arduino. Furthermore I do get an answer when I connect the Fona to my computer and use TTL com port and Putty.
However, I do not get an answer at any baud rate when using the TinyGSM library nor does the connection using Blynk work.
If 9600 baud works for the module over USB TTL, then should not the AT_debug also be able to successfully connect at that rate and report pass instead of fail?

GeorgianBorca, 2017-03-27:
I'm pretty shure that this is a connection issue.  Could you post the part of your code where is your software serial defined and the pins. Try to swap tham the RX TX pins in your code and you should pe good to go.

larsrengersen, 2017-03-27:
@GeorgianBorca Don't want to celebrate too early, but I think you are my hero!
> Try to swap tham the RX TX pins in your code
The example code says 
`SoftwareSerial SerialAT(50, 51); // RX, TX`
But this must be
`SoftwareSerial SerialAT(50, 51); // TX, RX`
After swapping the RX and TX wires, I got "Trying baud rate 115200... OK". 
[image]
Still some strange characters, but this is more than I've ever seen.
To be continued.

vshymanskyy, 2017-03-27:
Now SoftwareSerial @ 115200 tends to be unstable.
Either use HW serial, or switch to lower speed.

vshymanskyy, 2017-03-27:
@larsrengersen please tell us how it goes! ;)

GeorgianBorca, 2017-03-27:
Like @vshymanskyy said, use a lowe baudrate. I alway use 9600 or 4800.
The RX TX traces may not be length matched and may not be suitable for high baudrates and also software serial dose not work well at such high baudrates. Glad you made it so far.

larsrengersen, 2017-03-28:
Well, using baud rate 9600 results in a stable connection indeed.
Next I used the Diagnostics example. First two fails (1370 bytes instead of 1000), but then a pass as well.
Then I switched to the SIM800_SIM900 example again. Unfortunately still "cannot init".
As a "perhaps I am lucky" I added some parts of the Diagnostics example to the SIM800 example.
This is the serial monitor output:
`Initializing modem...
AT

OK
AT+CFUN=0

+CPIN: NOT READY

OK
AT+CFUN=1,1

OK
AT
AT

OK
AT&FZE0
AT&FZE0

OK
AT+CPIN?

ERROR
AT+CPIN?

+CPIN: READY

+CPIN: READY

OK
AT

OK
AT+CFUN=0

+CPIN: NOT READY

OK
AT+CFUN=1,1

OK
AT
AT

OK
AT&FZE0
AT&FZE0

OK
AT+CPIN?

ERROR
AT+CPIN?

+CPIN: READY

+CPIN: READY

OK
[16141] 
    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v0.4.6 on Arduino Mega

[16148] Modem init...
AT

OK
AT&FZE0
[17276] Cannot init`

So if it makes sense what I did in terms of combining and if I understand it correctly, the Fona does not respond to AT&FZE0 in a proper way.
Next I switched back to AT_debug and entered the command separately.
This was the output:
`Autodetecting baud rate
Trying baud rate 9600... OK
***********************************************************
 You can now send AT commands
 Enter "AT" (without quotes), and you should see "OK"
 If it doesn't work, select "Both NL & CR" in Serial Monitor
***********************************************************
`
Then I submitted AT&FZE0 and got:
`
ERROR
+CMTI: "ME",28
`
And now I'm lost in terms of what's next.... Hope you have a suggestion.

larsrengersen, 2017-03-28:
Don't know whether it is a proper fix, but I went into TinyGsmClientSIM800.h and changed line 181 from:
`    sendAT(GF("&F0"));  // Factory + Reset + Echo Off`
into
`    sendAT(GF("&FZE0"));  // Factory + Reset + Echo Off`
and now I am online and able to switch on a led using my iPhone.

larsrengersen, 2017-03-29:
[image]

larsrengersen, 2017-03-29:
Thanks for all your support, especially @GeorgianBorca

closed 2017-03-29 by larsrengersen

Linuxhombre, 2017-07-11:
Hi Everyone,

I'd love to be able to contribute.    

I'd like to personally add screen-based functionality (TFT) and then also LCD Shield and LCD + KeyPad Libraries.

Is this of interest?  If so, I'm happy to write the libraries based upon any foundational libraries the community-at-large with the **Fona 808** can find for a 'good start' so we don't have to recreate that wheel.  Please help there and I'll do the heavy-lifting.

Let's do it!  Go team FONA + 808 Makers.

-LinuxHombre

brvdg, 2017-07-11:
Hi,

take a look in my branche.
I added the GPS functions for the sim808 (2nd generation).

Greats Brun

Linuxhombre, 2017-10-26:
Hi Everyone,

I wanted to see if there's a good core library to build-out touch TFT screen functionalities and/or keypad/keypad shield/lcd integrated libraries yet?

Kind Regards,
Josh

Linuxhombre, 2017-10-26:
PS: I've got everything working perfectly otherwise on the FONA 808.   I just wanted to render an 'analog lcd / keypad' front-end controller library along with a more 'digital' one in which one could present to a screen.

Ideas?