vshymanskyy/TinyGSM #19: Shield connection

astro91, 2017-03-14:
Issue similar to the previous one (https://github.com/vshymanskyy/TinyGSM/issues/18).
I have a sim900 arduino shield,  similar to attached image.
[image]

When it is mounted on arduino uno, using this wonderful library, all works fine. But when I make connections with jumper wires it doesn't work. I use softwareserial on pins 7 and 8 at 9600 bauds and 9 for powering on/off the shield.
 
Manual AT commands works, but the AT_debug sketch doesn't find any baudrate, and cannot connect even with forced 9600 set. 

Should I connect other pins in addition to 7, 8 and 9 to make the library work?

Lots of thanks, also for the great job made with this library!
L.

vshymanskyy, 2017-03-23:
I think you have to contact your board vendor...
did you connect power wires (gnd, +5) correctly?

vshymanskyy, 2017-03-27:
Closed due to lack of interest

closed 2017-03-27 by vshymanskyy