vshymanskyy/TinyGSM #20: Automatic reconnection

Anydy, 2017-03-20:
Is it possible to automatically restore the link between the modem and the modem when the network or power is lost?
If so, how to implement it?

vshymanskyy, 2017-03-23:
You can restart your modem and reinit it for sure...

vshymanskyy, 2017-03-27:
Did you solve your problem?

Anydy, 2017-03-29:
I inserted a piece of code from the Setop into the LOOP
It is right? Or it is necessary as that on another?
So it reconnects when there is a power failure.
`void loop() {  
 if (mqtt.connected()) {
    mqtt.loop();
  } else {     
    // Reconnect every 10 seconds
    unsigned long t = millis();
    if (t - lastReconnectAttempt > 10000L) {
      lastReconnectAttempt = t;
     }
        if (mqttConnect()) {
        lastReconnectAttempt = 0;
          }
else if (!modem.gprsConnect(apn, user, pass)) {
       while (true);
}
    }
  }`

vshymanskyy, 2017-05-17:
Hi, so did you solve the problem completely? Are there any other problems?

vshymanskyy, 2017-09-02:
Closing due to lack of interest

closed 2017-09-02 by vshymanskyy