vshymanskyy/TinyGSM #21: ERROR LOOP after a short time

laserfreak1111, 2017-03-30:
I have a SIM900 connected to an arduino uno (Software Serial)  running the MQTT example.
I publish a message every 10 seconds. 
After about 3-5 Minutes the sketch will end up in a loop.
This also happens if i use Hardware Serial for the SIM900.
Can it be a bad serial connection? (Noise because of the GSM Module).
Or is there something wrong with the software.

Attached is a Debug log of the communication and the sktech

[testgsm2sktech.txt](https://github.com/vshymanskyy/TinyGSM/files/881055/testgsm2sktech.txt)

[debug.txt](https://github.com/vshymanskyy/TinyGSM/files/881052/debug.txt)

vshymanskyy, 2017-04-18:
Did you solve your issue?

laserfreak1111, 2017-04-18:
not yet. Maybe a bad communication between my atmega2560/ mega328p and the SIM900 / A6 . But i cant do more then connect RX to TX and TX to RX. I also hooked up a logic analyzer. But everything looks fine. It just looses the connection and keeps looping with that error :(.

Do you have any ideas ?

laserfreak1111, 2017-04-30:
Really ? . Ok i will try... I thought lower is better

holgerkoch, 2017-05-01:
But why it works with higher baudrate?
I think, this is not a definitive solution. Only the probability that it works will be better.

Best regards

Holger

gcardo, 2017-05-30:
i have same problem. infinit loop
arduino send "AT+CIPSTART=1" to sim800 and he responds "ERROR"

vshymanskyy, 2017-05-30:
could you try if solutions from https://github.com/vshymanskyy/TinyGSM/issues/30 work for you?

FabianInostroza, 2017-08-02:
This could be related with the issue #62  I submitted

vshymanskyy, 2017-09-02:
Please reopen if this still happens on master branch. Thank you!

closed 2017-09-02 by vshymanskyy