vshymanskyy/TinyGSM #25: WaitforNetwork doesn't actually wait for a network connection (SIM800)

SRGDamia1, 2017-04-06:
When I run waitForNetwork(), I expect that it will and keep trying until a connection is made.  But it seems to be making one attempt to get the network status and then just returning false.  In the code, I think the for loop of waitForNetwork() should actually be a while loop instead.  Am I doing something wrong?

```cpp
Serial.println(F("Waiting for network..."));  // For debugging
if (!_modem->waitForNetwork()){
  Serial.println("... Connection failed");  // For debugging
} else {
   _modem->gprsConnect(_APN, "", "");
}
```

SRGDamia1, 2017-04-06:
I'm sorry.  I think it's actually an extra return statement.  PR issued.

vshymanskyy, 2017-09-02:
Fixed. Thank you! See your PR for more comments.

closed 2017-09-02 by vshymanskyy