vshymanskyy/TinyGSM #27: A6 GSM module and

zhoufusong, 2017-04-09:


henkbredell, 2017-04-14:
hi zhoufusong, have you been able to sort this problem out?  I am having same issues, but only on SIM800l with an Arduino.  Extending the MQTT_SOCKET_TIMEOUT 30 did not help.

henkbredell, 2017-04-15:
I tried the higher baud rate of 115k baud and this solved the problem for me - as suggested by Holger in the other post.  For now this is working for me.  The GPRS unit I am using SIM800l has an auto bauding setting to make it very  easy to change the baudrate.

vshymanskyy, 2017-05-17:
@zhoufusong did it help?

psatya111, 2017-06-11:
I have the same issue with mqtt client with a6c gsm. any body resolved it?could you please share me the working code.my mail_id:psatya111@gmail.com.please help me.

vshymanskyy, 2017-09-11:
[image]
Just verified with A6 - everything works.

closed 2017-09-11 by vshymanskyy

amineazatour, 2017-12-12:
hi;
i had this problem and i resolve it by changing this command : mqtt.setServer(server, port);
by : PubSubClient (server, port, callback, client), to learn more this function you should visit this web site: https://pubsubclient.knolleary.net/api.html#PubSubClient2
and also and this is the most important , you have to disable the firewall of your PC if you use it as a mqtt broker. Good luck