vshymanskyy/TinyGSM #28: Hardware Serial Issue in Arduino Uno

darshit115, 2017-04-15:
Dear vshymanskyy,
        I am using your TinyGSM library with Arduino UNO R3. I am using GSM800L for the communication purpose.
As per Library Example if i use software serial with baud rate 9600 its working correct. but i want to use hardware serial instead of software serial than communication to broker gives fail at baud rate 9600 or lower baudrate and for baudrate >9600 its connecting with broker but after 2 to 3 minute it reconnect to broker and same thing continue at same interval. 

So could you please help me to solve this issue with hardware serial.

Thanks in Advance.
Darshit

vshymanskyy, 2017-04-18:
Did you solve your proble?

darshit115, 2017-04-18:
Dear vshymanskyy,

Still same issue i am facing.
Please help to solve this issue.

Thanks in advance,
Darshit

agcII, 2017-04-30:
Increase bauddrate. this resolve the problem for my.

vshymanskyy, 2017-05-16:
Thanks for the updates! Glad it worked.

closed 2017-05-16 by vshymanskyy