vshymanskyy/TinyGSM #29: HTTPS ?

zerblatt007, 2017-04-15:
Hi! This is a great library, but does it support HTTPS?

My SIM800L module seems to function with AT+ sequence in chapter "3.5.
HTTPS Get Method with HTTPS" from https://cdn-shop.adafruit.com/datasheets/SIM800+Series_SSL_Application+Note_V1.00.pdf

Should I be able to just switch SSL on by sending AT+HTTPSSL=1 to the module on some way and continue using this library?

vshymanskyy, 2017-04-18:
No, HTTPS stack is on the SIM800L, but you can not make SSL raw connections, AFAIK.
For HTTPS, you can use separate AT commands (it is not supported by this library).

mrWheel, 2017-08-07:
Is there an example of how-to send an URL over HTTPS???

tusharvb19, 2017-08-10:
I also need the HTTPS solution.
Help me.

vshymanskyy, 2017-09-02:
Neither of GSM modules that are currently supported, have ability to perform raw SSL connections via AT commands. HTTPS is supported by SIM800 series, but it's not how TinyGSM library is intended to be used. Some custom methods just for HTTPS could be developed, but I currently have no interest in this, sorry. Contributions are welcome.

vshymanskyy, 2017-09-12:
Oh well, after further investigation, I found that some SIM800 modules are able to perform raw SSL/TLS connections. These are good news, and added this into my feature table as "planned".

vshymanskyy, 2017-09-12:
Please try it out and tell me if it works for you.
On my `SIM800 R14.18` it works fine.
The other one, `SIM800 R13.08` reports that SSL is not supported.

vshymanskyy, 2017-09-12:
@zerblatt007 @mrWheel @tusharvb19 @xslim

closed 2017-09-12 by vshymanskyy

joseketos, 2017-10-10:
Hello @vshymanskyy,  does this library work with SIM800H (specifically adafruit feather fona)

currently it does not seem to detect the modem when defining:
 #define TINY_GSM_MODEM_SIM800

Modem: 
Waiting for network...AT+CREG?...

jameshowe, 2018-01-09:
@vshymanskyy I've tried with similar board (SIM800L R14.18) and I'm finding the success rate inconsistent. As per the [example](https://github.com/vshymanskyy/TinyGSM/blob/5d6a9afcf92e1f5601625194b2449e76bf652750/examples/HttpsClient/HttpsClient.ino), querying `cdn.rawgit.com` seems to work fine, however, querying https://httpbin.org/ip fails.

Also, I tried POST over SSL and it had about a 99% failure rate, I managed to get 1 successful POST (via https://requestb.in).

The response I see in the logs is always:

````
AT+CIPSTART=1,"TCP","httpbin.org",443

OK

1, CLOSE OK
````

Difficult to pin point exactly what the issue is, the request fails after about a second so presumably there is some sort of problem establishing the SSL connection. Oddly though, if I were to just GET https://google.com - that does work, consistently :/

Sort of suggests it may be an issue with certain SSL certificates (possibly some being too big). Any thoughts?

micw, 2018-06-23:
I got similar issues (inconsistency) with a SIM800L module and TLS. I debugged it using openssl as a server. The data is transmitted incredible slow (I can read it while it is transfered) and from time to time some characters are missing. So the success rate is <10% (depending on payload length).
I assume that it as to do something with buffering, maybe each character is encrypted and sent individually?

micw, 2018-06-23:
Update: I added some debug code to the library. It turned out that when I call

    client.print(GET_REQUEST);

the modemSend method is called for each single character.

micw, 2018-06-23:
I finally got it to work stable and fast by writing my request to a buffer and use client.write(buffer,len) rather than client.print(String)

vshymanskyy, 2018-06-23:
This is done by the underlying Arduino Core on some platforms. Please
report there.

wbmrcb, 2018-07-06:
@jameshowe using a dhparam.pem file with 1024bits works for me on nginx, for now...

jameshowe, 2018-07-06:
@wbmrcb in the end I ditched the library in favour of [Adafruit_FONA](https://github.com/adafruit/Adafruit_FONA) which seems a lot more stable.

wbmrcb, 2018-07-06:
Seems like this is a problem on server SSL configuration, `SIM800 R14.18` works fine on some servers only (Tried FONA library but I ended up with same results).
Downgrading to 1024bit DH params helps (but not good for security right ?).
Since SIM800L supports TLSv1.0 (not TLSv1.1 or TLSv1.2) could be a problem ?

micw, 2018-10-29:
@FabianInostroza I can barely imagine that the device is able to perform propper SSL cert validation because this would mean that it contains an regulary updated lost of trustworthy CA certificates as well as a real time clock.

ilgarbenli, 2023-12-05:
> This is done by the underlying Arduino Core on some platforms. Please report there.
> […](#)

how can I implement cert pem file in my code

commit 87af409c (2017-09-12): SSL/TLS support. Added GsmClientSecure. Fix #29