vshymanskyy/TinyGSM #30: BlynkClient example with A6 crashes after ~20hrs

marissv, 2017-05-06:
Blynk client works, but crashes after about 20hrs every time. After arduino reset works again for ~20 hrs, but never more.
Tried with Arduino ProMini, Arduino Mega. Last test with enabled debugg Serial 
```
[83811073] Connecting to blynk-cloud.com:8442
[83811144] <[02|00|01|00] AUTH_KEY
[83811202] Sent 65535/37
[83811316] Connecting to blynk-cloud.com:8442
```
Serial port dump between Arduino and A6:
```
AT+CIPSTART="TCP","blynk-cloud.com",8442
AT+CIPSEND=255,37
AT+CIPCLOSE=255
AT+CIPCLOSE=255
AT+CIPSTART="TCP","blynk-cloud.com",8442
AT+CIPSEND=255,37
AT+CIPCLOSE=255
AT+CIPCLOSE=255
```
A6 replies:
```
ERROR
ERROR
ERROR
ERROR
```
After Arduino reset (A6 stays powered), after modem.restart() command A6 "wakes up" and functions again.

vshymanskyy, 2017-05-10:
Yes I did not implement any complex error handling in example sketches.
Most modems do fail over time, and in general you'll have to reboot modem/trigger power to get it working again.
I would love to review PRs for such advanced examples. But probably it's not that easy - it would be a separate example for each modem type/model.

marissv, 2017-05-10:
Right now I'm going crazy with this. Seems like this arduino sample sketch sends a lot of AT commands actually not implemented in A6 module and not needed for TCP connection over GPRS.

marissv, 2017-05-13:
Spent a lot of time but no success yet. I'm not programmer, so it's hard to understand, what happens there.
I'm jam in these commands when blynk is crashed after some time:
```
AT+CIPSTART="TCP","blynk-cloud.com",8442
AT+CIPSEND=255,37
AT+CIPCLOSE=255
```
1) "blynk" understands that connection is lost and trying to connect again. I'll check if the A6 modem can reconnect again without restart (AT+RST=1) command.

2) Why blynk is trying to send something ower mux=255? If there any need for muxes more than 1?

3) A6 modem has 8 mux. From where programm takes mux 255?

What does this code (TinyGsmClientA6.h) do? I can't find ewerywhere. I asked because A6 modem answers CONNECT OK, IP CLOSE, IP INITIAL (for every mux 0:7)
```
  bool modemGetConnected(uint8_t mux) { //TODO mux?
    sendAT(GF("+CIPSTATUS"));
    int res = waitResponse(GF(",\"CONNECTED\""), GF(",\"CLOSED\""), GF(",\"CLOSING\""), GF(",\"INITIAL\""));
    waitResponse();
    return 1 == res;
  }
```
Is connection lost error handling should be in TINY_GSM_YIELD() function like +TCPCLOSED:?

What is the bool for GPRS connection lost to call connectNetwork() ?

marissv, 2017-05-24:
After some testing and debugging... success not so good.

A6 module is impossible to attach to GPRS network again without reset after crashed. (cant't turn on CGACT=1,1)
After crash  TinyGsmClient.h sock_connected=False and isConn=False
so far so good. :+1: Blynk knows it's dissconnected :)

How it is possible to push blynk to "start from beginning": connectNetwork(apn, user, pass)?
But it's really hard to get in these codes with classes and pointers so any help would be invaluable!

And maybe some little improvement:
Blynk.connectNetwork(apn, user, pass) calls modem.begin() so I suggest to change TinyGsmClient.h function:
```
  bool begin() {
    return init();
  }
  bool init() {
    if (!autoBaud()) {
      return false;
    }
    sendAT(GF("+RST=1"));
    delay(10000);  // some time to reset modem
    
    sendAT(GF("&FZE0"));  // Factory + Reset + Echo Off
    ....
```
I think It's safer to start connecting to network with restart. And there is no more need for modem.restart(); in main sketch.

marissv, 2017-05-30:
To stay compatible with lib, changed ino sketch.
Added lines to check, if Blynk.connected() and restarting modem, if not.
```
void loop()
{
  Blynk.run();
  if (!Blynk.connected()){
    Serial.println("Blynk was disconnected.... ");
    modem.restart();
    Blynk.begin(auth, modem, apn, user, pass);
    Serial.println("Modem was restarted and reconfigured... ");
  }
}
```

marissv, 2017-06-01:
Unfortunately, blynk was gone offline after about 7 days.

vshymanskyy, 2017-09-02:
Please try the latest release of the library, reopen if still have issues.

closed 2017-09-02 by vshymanskyy