vshymanskyy/TinyGSM #31: Library that just fail

XuTpuK, 2017-05-06:
Hi!
I have Mega2560, SIM800L and standard wiring:
SIM800L has separate strong power.
Mega TX -> SIM RX0
Mega RX -> SIM TX0

After sketch started, I open Serial Monitor and see rubbish (mostly empty squares). Maybe speed? At bottom of Monitor speed is 9600, obviously not a problem. Library which just fails...
I didn't run 'Diagnostics', because it requires downloading other lib, what is boring. Just stopped.

narzul, 2017-05-06:
That depends on what you have set the baud rate to be, the default when you open an example are SerialAT.begin(115200). I have used it with Mega2560 and Mega1280, with out any problems.
If you have pin number on the simcard you need to add it to the script or put the sim into a mobil and disable pin number.

vshymanskyy, 2017-05-10:
"Library which just fails... I didn't run 'Diagnostics', because it requires downloading other lib, what is boring. Just stopped."
I'm not blaming you... ;) Just find a better lib.

closed 2017-05-10 by vshymanskyy