vshymanskyy/TinyGSM #35: Example for a stable MQTT client

holgerkoch, 2017-05-15:
Hi, 
it seems, that the example MQTT client is not stable. Sometimes it works some hour, sometimes only few minutes.

Did somebody wrote a really stable MQTT client and can get me the code?

Here is my testcode, very close to the example code:

/**************************************************************
 *
 * For this example, you need to install PubSubClient library:
 *   https://github.com/knolleary/pubsubclient/releases/latest
 *   or from http://librarymanager/all#PubSubClient
 *
 * TinyGSM Getting Started guide:
 *   http://tiny.cc/tiny-gsm-readme
 *
 **************************************************************
 * Use Mosquitto client tools to work with MQTT
 *   Ubuntu/Linux: sudo apt-get install mosquitto-clients
 *   Windows:      https://mosquitto.org/download/
 *
 * Subscribe for messages:
 *   mosquitto_sub -h test.mosquitto.org -t GsmClientTest/init -t GsmClientTest/ledStatus -q 1
 * Toggle led:
 *   mosquitto_pub -h test.mosquitto.org -t GsmClientTest/led -q 1 -m "toggle"
 *
 * You can use Node-RED for wiring together MQTT-enabled devices
 *   https://nodered.org/
 * Also, take a look at these additional Node-RED modules:
 *   node-red-contrib-blynk-websockets
 *   node-red-dashboard
 *
 **************************************************************/

// Select your modem:
#define TINY_GSM_MODEM_SIM800
//#define TINY_GSM_MODEM_SIM900
//#define TINY_GSM_MODEM_A6
//#define TINY_GSM_MODEM_M590

#define TINY_GSM_RX_BUFFER 512

#include <TinyGsmClient.h>
#include <PubSubClient.h>

// Your GPRS credentials
// Leave empty, if missing user or pass
const char apn[]  = "web.vodafone.de";
const char user[] = "";
const char pass[] = "";

// Use Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial

// or Software Serial on Uno, Nano
//#include <SoftwareSerial.h>
//SoftwareSerial SerialAT(5, 4); // RX, TX

TinyGsm modem(SerialAT);
TinyGsmClient client(modem);
PubSubClient mqtt(client);

long lastReconnectAttempt = 0;

unsigned long lastSend = 0;

void setup() {
  // Set GSM module baud rate
  SerialAT.begin(115200);
  delay(3000);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  //Serial.println("Initializing modem...");
  modem.restart();

  //Serial.print("Waiting for network...");
  if (!modem.waitForNetwork()) {
    //Serial.println(" fail");
    while (true);
  }
  //Serial.println(" OK");

  //Serial.print("Connecting to ");
  //Serial.print(apn);
  if (!modem.gprsConnect(apn, user, pass)) {
    //Serial.println(" fail");
    while (true);
  }
  //Serial.println(" OK");

  // MQTT Broker setup
  mqtt.setServer("servername", 1883);
}

boolean mqttConnect() {
  if (!mqtt.connect("GsmClientTest", "username", "password")) {
    //Serial.println(" fail");
    //Serial.println(mqtt.state());
    return false;
  }
  //Serial.println(" OK");
  delay(10000);
  mqtt.publish("owntracks/wohnmobil/elnagh", "GsmClientTest started");
  return mqtt.connected();
}

void loop() {
  unsigned long time;

  
  char msgBuffer[20];
  
  if (mqtt.connected()) {
    mqtt.loop();
  } else {
    // Reconnect every 10 seconds
    unsigned long t = millis();
    if (t - lastReconnectAttempt > 10000L) {
      lastReconnectAttempt = t;
      if (mqttConnect()) {
        lastReconnectAttempt = 0;
      }
    }
  }
  if (millis()-lastSend>10000) {
    time = millis();
    mqtt.publish("owntracks/wohnmobil/elnagh", dtostrf(time, 6, 0, msgBuffer));
    lastSend=millis();
  }
}

best regards

Holger

himanshu-hearthacker, 2017-07-18:
i am having the same problem. it works for about 15 minutes, then stop sending data. Do you find any solution yet??

holgerkoch, 2017-07-18:
No! I tried much, but doesn't found a solution for this.

Holger

cimba007, 2017-07-19:
I am running the MQTT-Example now for ~1hour 40minutes (mqttspy toggles a variable every 10seconds). Can you tell me how you test your sketch? What are other clients publishing to the tinygsm-node and is the tinygsm-node publushing anything at all?

mcr-ksh, 2017-07-20:
have a look at issue #57. that might be that mqtt closes a socket and cant reopen it. enable Serial Debug via `#define TINY_GSM_DEBUG    Serial`

cimba007, 2017-07-20:
Is there a way to use TINY_GSM_DEBUG with softserial?

I am running fine with this setup 

```
#include <SoftwareSerial.h>
SoftwareSerial mySerial(2, 3); // RX, TX

// Set serial for debug console (to the Serial Monitor, speed 115200)
#define SerialMon mySerial

// Use Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial
```

But I get no real "verbose" messages

cimba007, 2017-07-21:
After reading https://github.com/vshymanskyy/TinyGSM/issues/57 I thought that modifying stop() to do the "AT+CIPCLOSE" if "AT+CIPCLOSE=MUX" fails with an error might be a workaround?

Not use to test this as I never got the error on mqtt on the wild.

mcr-ksh, 2017-07-24:
This is basically what I do manually:
```
client.stop();
SerialAT.println("AT+CIPCLOSE");
```

cimba007, 2017-07-24:
Where in the code do you call thus?

vshymanskyy, 2017-09-02:
Please try the latest release of the library. SIM800 was improved significantly

vshymanskyy, 2017-09-02:
Closing this for now, if experiencing issues with SIM800 - please reopen

closed 2017-09-02 by vshymanskyy