vshymanskyy/TinyGSM #39: ESP8266 & A& will not connect to network.

GeorgeDLake, 2017-05-25:
Hi, I cant get the lib to connect to my local network.
APN, UuserName, Password are correct.
It just will not pass:
  if (!modem.gprsConnect(apn, user, pass)) {
    Serial.println(" fail");
    delay(10000);
    return;
  }
any ideas?

GeorgeDLake, 2017-05-25:
I have added some more things in the diag script... but still cant find why it will not connect
Initializing modem...ATE0

OK
AT+RST=1
ATE0

OK
AT&FZE0

OK
AT+CMEE=0

OK
AT+CPIN?

+CPIN:READY

OK
Waiting for network...AT+CREG?

+CREG: 1,1

OK
 OK
Signal:AT+CSQ

+CSQ: 16,99

OK
16
Connecting to wap.tmovil.clAT+CIPSHUT

+CGREG: 1

COMMAND NO RESPONSE!
### Unhandled: +CGREG: 1

COMMAND NO RESPONSE!
AT+CREG?

+CREG: 1,1

OK
AT+CGACT?

+CGACT: 0,0

OK
AT+CGATT=1

OK
AT+CGATT=1

OK
AT+CGDCONT=1,"IP","wap.tmovil.cl"

OK
AT+CSTT="wap.tmovil.cl","wap","wap"

OK
AT+CGACT=1,1

OK
AT+CGATT=1

OK
AT+CIPMUX?

ERROR
 fail
************************
 Is GPRS enabled by network provider?
 Try checking your card balance.
************************

GeorgeDLake, 2017-05-27:
yes, all is ok from the card side. it seems to be an issue with the correct sequences of AT commands.

vshymanskyy, 2017-09-04:
I think Unhandled: +CGREG: 1 is no issue at all.
is "COMMAND NO RESPONSE!" your print or is it AT command rsp. from the modem?

vshymanskyy, 2017-09-10:
Please reopen if issue still happens.

closed 2017-09-10 by vshymanskyy