vshymanskyy/TinyGSM #40: Loss of connectivity doesn't force modem restart

anibald, 2017-06-04:
Hello!
Thanks for the great lib and forgive me if it isn´t the proper place to post this. The issue I'm having is related to a lost of connection with the GPRS network. I.e. the modem is still connected but I can't access any host. 

System setup:
Arduino Nano
Neoway M590e GPRS modem

Heres the log:

`Initializing modem...
[35153] 
    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v0.4.7 on Arduino Nano

[35161] Free RAM: 925
[35163] Modem init...
[35491] Connecting to network...
[54733] Network: TIM BRASIL
[54733] Connecting to timbrasil.br ...
[65097] Connected to GPRS
[69203] Connecting to blynk-cloud.com:8442
[293616] Connecting to blynk-cloud.com:8442
[320898] <[02|00|01|00] 019f2171b6e345f89c1c40ac5e255451
[328081] Connecting to blynk-cloud.com:8442
[328978] <[02|00|01|00] 019f2171b6e345f89c1c40ac5e255451
[330229] >[00|00|01|00|C8]
[330230] Ready (ping: 1122ms).
[330241] <[11|00|01|00]fver[00]0.4.7[00]h-beat[00]60[00]buff-in[00]256[00]dev[00]Arduino Nano[00]cpu[00]ATmega328P[00]con[00]SIM800[00]build[00]Jun  4 2017 13:17:30[00]
[332735] <[14|00|02|00|0B]vw[00]1[00]23.000
[332839] <[14|00|03|00|0B]vw[00]2[00]68.000
[333014] >[00|00|01|00|C8]
[392953] <[06|00|04|00|00]
[393335] <[14|00|05|00|0B]vw[00]1[00]23.000
[393445] <[14|00|06|00|0B]vw[00]2[00]68.000
[399041] <[06|00|07|00|00]
[431951] >[00|00|07|00|C8]
[453758] <[14|00|08|00|0B]vw[00]1[00]24.000
[454777] Cmd error
[456830] Cmd skipped:20
[459939] Connecting to blynk-cloud.com:8442
[691460] Cmd skipped:20
[691461] Cmd skipped:20
[694569] Connecting to blynk-cloud.com:8442
[926090] Cmd skipped:20
[926091] Cmd skipped:20
[929199] Connecting to blynk-cloud.com:8442
`

As you may see the modem initializes, the connection is established but then it loses the heartbeat and keeps trying to reconnect forever. A modem restart seems to solve the problem, but the code it self could´t been able to handle with this situation.
Looking on the library code I saw a private function "dnsIpQuery" which could be a good way to check if the connectivity is lost and force a modem restart but I haven't been able to understand how to use it. Dunno if it might help, but here's my sketch code:

`#define BLYNK_PRINT Serial    // Comment this out to disable prints and save space
#define BLYNK_DEBUG
// Default heartbeat interval for GSM is 60
// If you want override this value, uncomment and set this option:
//#define BLYNK_HEARTBEAT 30

#define TINY_GSM_MODEM_M590

#include <TinyGsmClient.h>
#include <BlynkSimpleSIM800.h>

#include <SimpleTimer.h>
#include <DHT.h>
#define DHTPIN 4 // O pino que estive conectado...
#define DHTTYPE DHT11 // DHT 11

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
const char auth[] = "********";

// Your GPRS credentials
// Leave empty, if missing user or pass
const char apn[]  = "timbrasil.br";
const char user[] = "tim";
const char pass[] = "tim";

// Hardware Serial on Mega, Leonardo, Micro
//#define SerialAT Serial1

// or Software Serial on Uno, Nano
#include <SoftwareSerial.h>
SoftwareSerial SerialAT(2, 3); // RX, TX

TinyGsm modem(SerialAT);
DHT dht(DHTPIN, DHTTYPE); //Da um nome para o Sensor
SimpleTimer timer; // define o objeto timer

void setup()
{
  // Set console baud rate
  Serial.begin(115200);
  delay(10);

  // Set GSM module baud rate
  SerialAT.begin(9600);
  delay(3000);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  Serial.println("Initializing modem...");
  modem.restart();

  // Unlock your SIM card with a PIN
  //modem.simUnlock("1234");

  Blynk.begin(auth, modem, apn, user, pass);
  dht.begin();
  delay(2000);
  sendUpdtTempHum();
  timer.setInterval(60000, sendUpdtTempHum);
}

void loop()
{
  while(modem.dnsIpQuery("blynk-cloud.com")) {
    Blynk.run();
    timer.run();
  }
  modem.restart();
 }

void sendUpdtTempHum()
{
  float humid = dht.readHumidity();
  float temp = dht.readTemperature();
//  Serial.print("Temp: ");
//  Serial.println(temp);
//  Serial.print("Humid: ");
//  Serial.println(humid);
  Blynk.virtualWrite(1, temp);
  Blynk.virtualWrite(2, humid);
}`

anibald, 2017-06-06:
Hello again,

Made some advances in handling the losses of connection, my loop function now looks like this:

```
void loop()
{
    Blynk.run();
    timer.run();
    int timestamp = millis();
    if (!Blynk.connected() || !modem.gprsConnect(apn, user, pass)) {
      Serial.println(sprintf("[%d] Connectivity is gone", timestamp));
      modem.restart();
      Blynk.begin(auth, modem, apn, user, pass);
      Serial.println("Modem was restarted and reconfigured... ");
    }    
}

```

It's now forcing reboot more frequently but I still get some situation where Blynk.connected and modem.gprsConnect returns TRUE but in fact the serial shows the situation below, where no error is given and that just stops after a hard reset.

Is there some option to put TinyGSM in debug mode?

```
[53355356] Heartbeat timeout: 53355356, 53269437, 53344450
Connectivity is gone... 
[53386370] 
    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v0.4.7 on Arduino Nano

[53386378] Free RAM: 779
[53386380] Modem init...
[53386703] Connecting to network...
[53435635] Network: TIM BRASIL
[53435636] Connecting to tim.br ...
[53445969] Connected to GPRS
[53450073] Connecting to blynk-cloud.com:8442
[53685421] Connecting to blynk-cloud.com:8442
[53920768] Connecting to blynk-cloud.com:8442
[54156113] Connecting to blynk-cloud.com:8442
[54391458] Connecting to blynk-cloud.com:8442
Initializing modem...

```

vshymanskyy, 2017-06-06:
Just to be sure, verify that your code does what you wanted to. It looks invalid to me.
also, it is recommended to reset your nodem using reset/enable pin of modem or triger power directly.

closed 2017-06-06 by vshymanskyy