vshymanskyy/TinyGSM #46: Problem GPS with SIM808

ilbonch, 2017-06-14:
Hi, I have a dfrobot sim808 shield   www.dfrobot.com/product-1469.html
Thanks to brvdg I got data via the "getGPSraw" function.  www.github.com/vshymanskyy/TinyGSM/pull/34 
I wanted to know how to get latitude and longitude through the call "getGPS".
Otherwise how can I get extrapolation from getGPSraw only latitude and longitude?

Thank

brvdg, 2017-06-22:
Here is an example how I use it.

`    gps_fixstatus = modem.getGPS(&gps_latitude, &gps_longitude, &gps_speed, &gps_altitude, &gps_view_satellites, &gps_used_satellites);
    if ( gps_fixstatus ) {
      
      DEBUG_PRINT(F("#GPS Location: LAT: "));
      DEBUG_PRINT(gps_latitude);
      DEBUG_PRINT(F(" LONG: "));
      DEBUG_PRINT(gps_longitude);
      DEBUG_PRINT(F(" SPEED: "));
      DEBUG_PRINT(gps_speed);
      DEBUG_PRINT(F(" ALTITUDE: "));
      DEBUG_PRINT(gps_altitude);
      DEBUG_PRINT(F(" USED SATELITES: "));
      DEBUG_PRINT(gps_view_satellites);
      DEBUG_PRINT(F(" VIEWED STELITES: "));
      DEBUG_PRINTLN(gps_used_satellites);

      int year, month, day, hour, minute, second;
      if ( modem.getGPSTime(&gps_year, &gps_month, &gps_day, &gps_hour, &gps_minute, &gps_second) ) {
        //Sync time if it's diffrent.
        set_time(gps_year, gps_month, gps_day, gps_hour, gps_minute, gps_second);
      }
      
    }`

I hope it helps.

Greats

ilbonch, 2017-07-01:
Thank you very much!!! With your help now it all works !!!!

closed 2017-09-02 by vshymanskyy

Sbeemapp, 2018-07-28:
Hola alguien sabe como conectar este modulo a un servidor  para almacenar datos en MySQL

sebastianfelipe, 2018-09-09:
Hi, I have a doubt. The result is shown with .2f precision, is there possible to show with a big number? Thanks.

brvdg, 2018-09-09:
Hi,

Sorry, I don't understand what exactly do you mean.
The output is the same what you get from the module.

sebastianfelipe, 2018-09-12:
No, is not the same. When I use the get gps raw function it only returns me floats with only 2 values of precision. I modified the library to return String and in that way I could get everything. There is a bug in that function. I think that use a String instead of float and int types is better. After that it could be parsed, but doing it before, it has problemas like precision lose. I compiled it on OSX 10.11.

brvdg, 2018-09-12:
I can't confirm it.
In my project I get a float with a precision of 10. Also I compile my project on OSX.

Pleas try the function getGPSraw(). This returns the string which you get directly from the SIM808.

There is nothing in the lib which limit the precision on the float.

sebastianfelipe, 2018-09-12:
I thought the same, but in my case it happens. when the toFloat() method is applied to the result, then the referenced latitude, longitude and altitude are drastically changed. I don't know why it happens. I think I'm gonna try to use the getGPSraw() instead of getGPS :/

brvdg, 2018-09-13:
To see the position I think you use a print function.
How do you use it?