vshymanskyy/TinyGSM #47: I am starting that this is a bug instead.

noctilucentsky, 2017-06-14:
#include <TinyGsmClient.h>
#include <StreamDebugger.h>

// Set serial for debug console (to the Serial Monitor, speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
//#define SerialAT Serial1

// or Software Serial on Uno, Nano
#include <SoftwareSerial.h>
SoftwareSerial SerialAT(10, 11); // TX, RX

StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
//TinyGsm modem(Serial);
TinyGsmClient client(modem);

void setup()
{
  // Set console baud rate
  SerialMon.begin(38400);
  delay(10);
  SerialMon.println("1");

  // Set GSM module baud rate
  SerialAT.begin(57600);
  SerialMon.println("2");
  delay(3000);
  SerialMon.println("3");

  if (!modem.init())
  {
    SerialMon.println("4");
    return;
  }
    SerialMon.println("5");
  modem.sendSMS("0769879696", "Text here");
  SerialMon.println("6");
}

Does not send an SMS, but prints a weird character instead after the text supposedly to be sent.

noctilucentsky, 2017-06-16:
Are there any ways to wait for the SIM800 to completely initialize and check for network connectivity before starting to send commands?

It would definitely be helpful.

vshymanskyy, 2017-06-19:
This is not a weird character. This is how SIM800 should work.

vshymanskyy, 2017-09-02:
Closed due to lack of interest.

closed 2017-09-02 by vshymanskyy