vshymanskyy/TinyGSM #49: modem.getBattVoltage() cause lockup

jalkestrup, 2017-06-18:
Hi there

Thanks for a nice and compact GSM library. I've recently tried adding the getBattVoltage to my sketch (operating on a SIM800H) and noticed that calling this function cause my code to hang roughly every 5-20th time it's being called (a 5x8 sec watchdog interrupt alert me that I'm hanging).

I'm sure that it is the function which cause these crashes, but it might be the product of factors I don't consider, when the code hangs I never reach the "vBat OK" print - and if I comment out the modem.getBattVoltage() the code runs without problems for hours. 

Any ideas on how to fix this or what it might be caused by?

```
  payload += ",\"SG\":" ; //Sleep Gateway
  payload += sleepTime; // append sleep time of node
  Serial.print(F("Sleep OK: "));
  delay(10);
  //Use: float vBatt = modem.getBattVoltage() / 1000.0;
  float vBatt = modem.getBattVoltage() / 1000.0 ;
  payload += ",\"BG\":" ;//
  payload += vBatt; // append battery of gateway
  Serial.print(F("vBat OK: "));
  payload += "}}";
  Serial.print(F("Sending payload: "));
  Serial.println(payload);
  delay(50);
```
Kind regards
Jesper

vshymanskyy, 2017-07-23:
You need to enable Debug Serial and see how your modem responds to these commands.

vshymanskyy, 2017-09-11:
Just try on the latest library from master. I have tested it like this:
```
  for(;;) {
    float vBatt = modem.getBattVoltage() / 1000.0 ;
    Serial.println(vBatt);
  }
```
And it looks really stable

closed 2017-09-11 by vshymanskyy