vshymanskyy/TinyGSM #55: Diagnostics from tools "### Unhandled:" or "nothing on read()

mcr-ksh, 2017-07-17:
Hi,

two issues with the same result.

using the A6 modem on a A20 with ESP8285 and SoftwareSerial on 9600 baud I receive 
"### Unhandled: OK" or "no content at all"

I use the 
https://github.com/arduino-libraries/ArduinoHttpClient/
and receive a timeout.

This scatch will lead lead to the "###Unhandled:" debug statements while the request is seen on the server side. "###Unhandled:" will appear on the server response upon +CIPRCV.
If "###Unhandled:" does not appear, you can see from the debug log that CIPRCV receives the data but the gsmclient.read() API does not receive a single byte.

```
  
  if (!gsmclient.connect(SERVER, HTTP_PORT)) {
    delay(10000);
    return ret;
  }

  // Make a HTTP GET request:
  gsmclient.print(String("GET ") + url + " HTTP/1.1\r\n");
  gsmclient.print(String("Host: ") + SERVER + "\r\n");
  gsmclient.print("Connection: close\r\n\r\n");

  unsigned long timeout = millis();
  while (gsmclient.connected() && millis() - timeout < 10000L) {
    // Print available data
    while (gsmclient.available()) {
      char c = gsmclient.read();
      Serial.print(c);
      timeout = millis();
    }
  }

  gsmclient.stop();
```
```
chksum Initializing modem...ATE0
ATE0
ATE0
ATE0
ATE0
ATE0
ATE0

OK
AT+RST=1
ATE0
ATE0
ATE0
ATE0
ATE0
ATE0
ATE0
ATE0
ATE0
ATE0
ATE0
ATE0
ATE0
ATE0
ATE0
ATE0
ATE0

OK
AT&FZE0

OK
AT+CMEE=0

OK
AT+CPIN?

ERROR
AT+CPIN?

+CPIN:SIM PIN

OK
AT+CPIN="6450"

OK
AT+CPIN?

^STN: 37

^CINIT: 4, 8192AT+CPIN?
CINIT: 32, 0, 0

+CREG: 0
AT+CPIN?

+CPIN:READY

+CPIN:READY

OK
Waiting for network...AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,10

OK
AT+CREG?

+CREG: 1,1

OK
 OK
Connecting to internetAT+CIPSHUT

ERROR
AT+CGDCONT=1,"IP","internet"

OK
AT+CSTT="internet","",""

OK
AT+CGACT=1,1

+CTZV:17/07/17,22:22:35,+02

OK
AT+CGATT=1

OK
AT+CIPMUX=1

OK
 OK
Connecting to cdn.rawgit.comAT+CIPSTART="TCP","cdn.rawgit.com",80

+CIPNUM:0

CONNECT OK

OK
 OK
AT+CIPSEND=0,65

>GET /vshymanskyy/tinygsm/master/extras/test_simple.txt HTTP/1.1

OK
AT+CIPSEND=0,22

>Host: cdn.rawgit.com

OK
AT+CIPSEND=0,21

>Connection: close

OK

+CIPRCV:0,402,HTTP/1.1 200 OK
Date: Mon, 17 Jul 2017 22:22:43 GMT
Content-Type: text/plain;charset=utf-8
Transfer-Encoding: chunked
Connection: close
X-Content-Type-Options: nosniff
X-Robots-Tag: none
Access-Control-Allow-Origin: *
ETag: "675136a9bcc9b879e0ebcfbee3eb7bc7745c2e27"
Cache-Control: max-age=86400
Vary: Accept-Encoding
RawGit-Cache-Status: HIT
Server: NetDNA-cache/2.2
X-Cache: HIT

3e8

 +CIPRCV:0,402,01 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
02 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
03 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
04 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
05

 +CIPRCV:0,402, #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
06 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
07 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
08 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
09 #

 +CIPRCV:0,203,$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
10 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz

0

 
+TCPCLOSED:0

OK
AT+CIPCLOSE=0

ERROR
Server disconnected
AT+CIPSHUT

ERROR
GPRS disconnected

************************
 Received: 0 bytes
 Test:     FAIL
************************
```

vshymanskyy, 2017-07-23:
It looks like there is some extra character "�" which clobbers +CIPRCV .. Need to check if we can work-around it somehow...

vshymanskyy, 2017-09-10:
Could you try it on latest master branch? Shold be resolved. Closing for now.

closed 2017-09-10 by vshymanskyy