vshymanskyy/TinyGSM #56: ### Buffer overflow

mcr-ksh, 2017-07-17:
Hi,

so at some point internally it seems the response causes to exceed TINY_GSM_RX_BUFFER 

`+CIPRCV:1,349,### Buffer overflow:  349 -> 255`

if I increase the buffer by
`#define TINY_GSM_RX_BUFFER  512`

its ok.
`+CIPRCV:1,349,### Got:  349 -> 511`

how can I receive bigger buffers if the HTTP Response is huge content?

vshymanskyy, 2017-07-22:
Unfortunately I didn't see an option to limit the size of buffer. That's why It may require much bigger buffers (and more expensive main controller).
If you find any ideas/ways to workaround this limitation, please re-open the issue.

closed 2017-07-22 by vshymanskyy

yi-ge, 2017-10-22:
I have the Same Problem.

SRGDamia1, 2017-10-23:
The ESP8266 also has this problem.  I haven't seen any way to decrease the receive buffer for it, either.