vshymanskyy/TinyGSM #57: AT+CIPCLOSE=ID ERROR upon AT+CIPSEND failure

mcr-ksh, 2017-07-18:
apparently the lib does not handle the close properly when a sending failure occurred or any other error occurs (dont know what exactly happends).

Closing the socket aka httpclient->stop(); will trigger the AT+CIPCLOSE=mux, which results in an ERR, especially if the CIPSTART was not successful. Thus the successive CIPSTART can only be closed via "CIPCLOSE" without the "=1".

My assumption:
AT+IPMUX=1 was set at some point. However, there is only one connection opened 
and closing that connection will cause an ERROR when calling AT+CIPCLOSE=1.
Every other further connection is then blocked. only a "AT+CIPCLOSE", without the =1, does the trick.

vshymanskyy, 2017-07-19:
you didn't specify any details of your hardware setup

FabianInostroza, 2017-08-05:
Hi @mcr-ksh, look at #62, I think this is the same problem, unfortunately I don't have an A6 modem to test

vshymanskyy, 2017-09-12:
This bug is confirmed

xvinny, 2020-03-18:
Any advance on this?

SRGDamia1, 2020-03-18:
Not that I'm aware of; I don't have one of these modules to test.