vshymanskyy/TinyGSM #58: [SIM908][DFROBOT Shield][Arduino Mega][GSM] problem Initializing modem

antoniofcano, 2017-07-18:
Hello, 

I'd like to use the TinyGSM library but until now for me is something impossible. 

First of all, I've been able to connect using the "AT Command Tester" app and the module works fine but when I try to use the TinyGSM with the Diagnostic I get this output:

> Initializing modem...
> fail
> ************************
> Is your modem connected properly?
> Is your serial speed (baud rate) correct?
> Is your modem powered on?
> Do you use a good, stable power source?
> Try useing File -> Examples -> TinyGSM -> tools -> AT_Debug to find correct configuration
> ************************

Following the instructions from [dfrobot wiki](https://www.dfrobot.com/wiki/index.php/GPS/GPRS/GSM_Module_V3.0_(SKU:TEL0051)#Drive_the_module_via_USB_port_.28AT_command.29) for use the AT Command Tester app:

1. Shield switchs S1: Prog and S2: Arduino
2. Loaded sketch
 ```
void setup() {
   //Init the driver pins for GSM function
   pinMode(3, OUTPUT);
   pinMode(4, OUTPUT);
   pinMode(5, OUTPUT);
   //Output GSM Timing
   digitalWrite(5, HIGH);
   delay(1500);
   digitalWrite(5, LOW);
}

void loop() {
   digitalWrite(3, HIGH); //disable GSM TX、RX
   digitalWrite(4, HIGH); //disable GPS TX、RX
}
```
3. Power off and put Shield switchs S1: Comm and S2: Arduino
4. Power on and connect the AT Command Tester with 9600 bauds
5. Send AT commands as normal and working fine.

Now If I try to load the Diagnostics sketch of the TinyGSM library  it doesn't work, and I'm totally lost.

Before upload the code I modified a little bit:

1. Select the modem: #define TINY_GSM_MODEM_SIM900
2. Adapt the baud setup:
```
void setup() {
  // Set console baud rate
  SerialMon.begin(9600);
  delay(10);

  // Set GSM module baud rate
  SerialAT.begin(9600);
  delay(3000);
}
```

The arduino mega is connected to an external power supply of 12v and 2.5A. And I follow this [instructions](http://www.raviyp.com/embedded/183-sim900-sim908-gsm-module-autobaud-solution) for change the Auto baud setup of the module to 9600.

Any ideas? 

I know that place is for issues about the library and my question is support question, but I don't know where I have to go. A lot of thanks!!

vshymanskyy, 2017-07-23:
Could you please show the full trace of AT commands, how did you manage to connect to the network manually? Also, the trace of commands that you get with the Diagnostics sketch?

vshymanskyy, 2017-09-02:
Closing due to lack of interest

closed 2017-09-02 by vshymanskyy