vshymanskyy/TinyGSM #59: problem when sending email..

herryWidnyana, 2017-07-19:
I want to create a project that can send email directly from arduino via at command. I've tried adding a function called sendEmail. Email successfully sent but the body/content of the message is empty.

Here is my testcode..
/********************************************************************************************************/
 bool sendEmail(const String& smtp_server, const String& smtp_port, const String& smtp_username, const String& smtp_password, const String& sender_address, const String& sender_name, const String& to_address, const String& to_name, const String& subject, const String& body, const String& apn, const String& username, const String& pass)
{
	sendAT(GF("+SMTPSRV=\""), smtp_server, GF("\",\""), smtp_port, '"');
	waitResponse();
	
	sendAT(GF("+SMTPAUTH=1,\""), smtp_username, GF("\",\""), smtp_password, GF("\""));
	waitResponse();
	
	sendAT(GF("+SMTPFROM=\""), sender_address, GF("\",\""), sender_name, GF("\""));
	waitResponse();
	
	sendAT(GF("+SMTPRCPT=1,0,\""), to_address, GF("\",\""), to_name, GF("\""));
	waitResponse();
	
	sendAT(GF("+SMTPSUB=\""), subject, GF("\""));
	waitResponse();
	
	sendAT(GF("+SMTPBODY=\""), body, GF("\""));
        waitResponse();
  
	sendAT(GF("+CGSOCKCONT=1,\"IP\",\""), apn, '"');
	waitResponse();
	
	sendAT(GF("+CSOCKAUTH=1,1,\""), username, GF("\",\""), pass, GF("\""));
	waitResponse();
	
	sendAT(GF("+SMTPSEND"));
	if (waitResponse(GF(GSM_NL "+SMTP: SUCCESS"))!= 1)
	{
		//return false;
		return waitResponse() == 0;
	}
	else
	{
		//return true
		return waitResponse() == 1;
	}
  }
/********************************************************************************************************/

vshymanskyy, 2017-07-23:
@herryWidnyana Let me invite you to or new TinyGSM developer chat:
https://gitter.im/tinygsm/Developers?utm_source=share-link&utm_medium=link&utm_campaign=share-link

vshymanskyy, 2017-07-23:
Which hardware did you use?

vshymanskyy, 2017-07-23:
Maybe you should dig into encodings... Did you manage to send an email using AT commands manually?

herryWidnyana, 2017-07-23:
Is there anything wrong with my coding ?. Email was sent, and I received the message, but the contents/body of message is empty.

I've tried using this....

sendAT(GF("+SMTPBODY"));
waitResponse(GF(">"));
stream.print(body);
stream.write((char)0x1A);
waitResponse();

but still empty..

vshymanskyy, 2017-07-23:
Did you manage to send an email using AT commands manually? Which hardware did you use?

herryWidnyana, 2017-07-23:
I am using adafruit fona 808 and arduino mega.

vshymanskyy, 2017-09-02:
Closing this for now, maybe at some point in time we will consider to add email function to the library.

closed 2017-09-02 by vshymanskyy