vshymanskyy/TinyGSM #60: using AT_Debug I can get a connection. Diagnostic on the other hand... SIM900 software serial

orotemo, 2017-07-20:
I get this:
(I added some Serial output: ______ signifies we got a response. -------- signifies timeout in `waitResponse`)

```bash
Initializing modem...AT
### AT: 

OK
______
AT+CFUN=0
### AT: +CFUN=0

OK
______
AT+CFUN=1,1
### AT: +CFUN=1,1

+CPIN: NOT READY
�����------
______
### Unhandled: +CPIN: NOT READY
�����
 fail
************************
 Is your modem connected properly?
 Is your serial speed (baud rate) correct?
 Is your modem powered on?
 Do you use a good, stable power source?
 Try useing File -> Examples -> TinyGSM -> tools -> AT_Debug to find correct configuration
************************
Waiting for network...AT+CREG?
### AT: +CREG?

+CREG:______
 0,1

OK
______
 OK
Connecting to net.hotmAT+CIPSHUT
### AT: +CIPSHUT

SHUT OK
______
AT+SAPBR=3,1,"Contype","GPRS"
### AT: +SAPBR=3,1,"Contype","GPRS"

OK
______
AT+SAPBR=3,1,"APN","net.hotm"
### AT: +SAPBR=3,1,"APN"," net.hotm "

OK
______
AT+CGDCONT=1,"IP","net.hotm"
### AT: +CGDCONT=1,"IP"," net.hotm "

OK
______
AT+CGACT=1,1
### AT: +CGACT=1,1

OK
______
AT+SAPBR=1,1
### AT: +SAPBR=1,1

ERROR
______
AT+SAPBR=2,1
### AT: +SAPBR=2,1

+SAPBR: 1,3,"0.0.0.0"

OK
______
AT+CGATT=1
### AT: +CGATT=1

OK
______
AT+CIPMUX=1
### AT: +CIPMUX=1

OK
______
AT+CIPQSEND=1
### AT: +CIPQSEND=1

OK
______
AT+CIPRXGET=1
### AT: +CIPRXGET=1

OK
______
AT+CSTT="net.hotm","",""
### AT: +CSTT=" net.hotm ","  ","  "

OK
______
AT+CIICR
### AT: +CIICR

+PDP: DEACT

ERROR
______
```

Any recommendations? your library does seem the most user friendly one, and I'd like to use it in my project.

Thanks in advance!

vshymanskyy, 2017-07-23:
Maybe you should just disable PIN code on your SIM card?

orotemo, 2017-07-23:
running `AT_Debug` tool, I get this:

```
AT+CPIN?
+CPIN: READY

OK
```

So I'm not sure what you mean by 'disable'. can you please elaborate. thanks again...

vshymanskyy, 2017-07-23:
Did you try to factory-reset your module?

orotemo, 2017-07-24:
Yes I did. I suspected the `CIICR`. something with pod maybe. what do you think? Did it work for you on sim900 "out of the box"?

orotemo, 2017-07-25:
sorry, typo - pod --> `PDP` 
maybe this `+PDP: DEACT` at the end is the source of the trouble? anyone else successful at getting this library to function on SIM900 without modifications (other then un/comenting the correct  `#define` statements of course)??

zerblatt007, 2017-07-25:
I have gotten the same +PDP: DEACT problem on 4 different SIM800 modules.
When I get a new one, it works for 1-2 days sending like it should, then it will not register to the network anymore. Registration denied..
That is when I get this +PDP: DEACT.

I have not found the source of this problem. The last module has a battery input so the it can not be the power supply.

vshymanskyy, 2017-09-02:
Please try the latest release of the library.

vshymanskyy, 2017-09-12:
Is it still reproducible with latest release? If yes, please reopen this issue.

closed 2017-09-12 by vshymanskyy