vshymanskyy/TinyGSM #61: [A6] downloading big files

mcr-ksh, 2017-07-20:
Hi,

so in your "FileDownload" example you say:
" * ATTENTION! Downloading big files requires quite a lot
 * of knowledge - so this is for more experienced developers."

So basically this is what im trying and I find the library faulty because it seems not be able to handle the received data properly as related to #55 and #56. Data is simply skipped by the library if it cant be stored. "### Unhandeled". 

Could you give a clue what other knowledge you are referring to?

Thx

vshymanskyy, 2017-07-23:
@mcr-ksh Let me invite you to our new developer forum: https://gitter.im/tinygsm/Developers?utm_source=share-link&utm_medium=link&utm_campaign=share-link

BTW, The knowledge I referred here - is exactly this: how the library works with your hardware, AT protocol, buffering, etc. + how server sends the data (how fast, how much bytes at a time) + how your microcontroller processes the data (does it have enough RAM to store chunks)..

closed 2017-07-23 by vshymanskyy

shreyaspaqs, 2018-12-19:
Hi, am trying download 700kb bin file from tiny gsm file download , am using ESP32 module for interface.but in example its written as for downloading large file need have knowledge on internals of tinygsm. please guide me on this. Regards.

taufikterdidik, 2020-10-05:
I almost success, only need more fixing/tuning.. same with WebClientExample, if readLength is less than ContentLengh, retry with same code except add this line
client.print(String("Range: bytes=") + (readLength-1) + "-" + (contentLength-1) + "\r\n"); //need fixing
so it continue download..

taufikterdidik, 2022-02-13:
> > I almost success, only need more fixing/tuning.. same with WebClientExample, if readLength is less than ContentLengh, retry with same code except add this line client.print(String("Range: bytes=") + (readLength-1) + "-" + (contentLength-1) + "\r\n"); //need fixing so it continue download..
> 
> @taufikterdidik Hi! Did this method work for you to download big files? Could you explain in more detail for the line: `client.print(String("Range: bytes=") + (readLength-1) + "-" + (contentLength-1) + "\r\n"); //need fixing`

it mean to download remaining file. If our downloaded file is less than total length of file (that to be downloaded), we resume download from last success bit to end of file

zark-zeugan, 2022-02-13:
> it mean to download remaining file. If our downloaded file is less than total length of file (that to be downloaded), we resume download from last success bit to end of file

But where to put this?