vshymanskyy/TinyGSM #62: [SIM868] library doesn't receive "CLOSED, 1" message from modem

FabianInostroza, 2017-08-02:
I'm testing this library and the mqtt example using a SIM868 module. It works for some time and then it stop working. I have logged the messages between the microcontroller and the modem and when the problem starts when the modem sends a "CLOSED, 1" string, I added some messages to the library to see what happens and this message was not being processed by the waitResponse function.

Because the connection was closed by the server (or the modem) there is ping timeout and the library tries to close the connection but this generate an error in the modem because the connection was already closed and the library is unable to recover from this state.

I have tested this with an arduino due and with an stm32f103, so I don't think this is a serial driver issue.

FabianInostroza, 2017-08-05:
Hi @vshymanskyy 

I think I found the cause of the problem.

Sometimes the modem receives data on a socket just after the microcontroller has sent data in [modemSend()](https://github.com/vshymanskyy/TinyGSM/blob/e90fc43a89d3c5bc3bafa06408961746a129658c/TinyGsmClientSIM800.h#L544), then the microcontroller waits for "DATA ACCEPTED" but before that a "+CIPRXGET" string was received so the 
waitResponse() method set the gotData flag. After returning from the waitResponse() method a call to modemGetAvailable() is made (because of the gotData flag) and this method removes part of the "DATA ACCEPTED" sentence, specifically the mux and number of bytes.
When we return to modemSend() the microcontroller waits for the mux and number of bytes but they were removed by the call to waitResponse() made in modemGetAvailable(), so modemSend() blocks until the modem send a comma and newline character. This causes the connection to timeout and the function receives "1" as mux and "CLOSED" as number of bytes (0 after conversion to int), this is interpreted as an error by the higher level function. For more details see the attached log files-
[log_arduino_dbg.txt](https://github.com/vshymanskyy/TinyGSM/files/1202641/log_arduino_dbg.txt)
[log_modem_tx.txt](https://github.com/vshymanskyy/TinyGSM/files/1202642/log_modem_tx.txt)

Having non blocking streamSkipUntil() and stream.readStringUntil() would have prevented the connection timeout but it would have not solved the real issue.

vshymanskyy, 2017-09-02:
Please reopen if this still happens on master branch. Thank you!

closed 2017-09-02 by vshymanskyy