vshymanskyy/TinyGSM #63: Timeout Issue in MQTTClient

architmuchhal12, 2017-08-02:
Hello,

I am using ESP32 Arduino with Adafruti FONA 808 to connect to the MQTT Server. I have modified the **MQTTClient.ino** accordingly.

I am using hardware serial at 115200 baud rate. The issue I am facing is the serial monitor which displays the reason for no connection as **(-4)** which is due to Timeout Value. I also tried increasing the timeout value in the library but with no success.

Please suggest me a solution!
Thanks!

**Init Code:**
```
// Select your modem:
#define TINY_GSM_MODEM_SIM800
//#define TINY_GSM_MODEM_SIM900
//#define TINY_GSM_MODEM_A6
//#define TINY_GSM_MODEM_M590

#include <TinyGsmClient.h>
#include <PubSubClient.h>
#include "Adafruit_FONA.h"

// Default pins for Feather 32u4 FONA
#define FONA_RST 25
HardwareSerial fonaSerial(1);
Adafruit_FONA fona = Adafruit_FONA(FONA_RST);

// Your GPRS credentials
// Leave empty, if missing user or pass
const char apn[]  = "wholesale";
const char user[] = "0";
const char pass[] = "0";

TinyGsm modem(fonaSerial);
TinyGsmClient client(modem);
PubSubClient mqtt(client);

const char* broker = "iot.wabashnational.com";
const char* userName = "******";
```

Serial Monitor Output:
```
Attempting to open comm with ATs
	---> AT
	<--- BU
	---> AT
	<--- ÿ
	---> AT
	<--- 
	---> AT
	<--- ÿ
	---> AT
	<--- AT
	---> AT
	<--- AT
	---> ATE0
	<--- ATE0
	---> ATE0
	<--- OK
	---> AT+CVHU=0
	<--- OK
	---> ATI
	<--- SIM808 R14.18

OK

FONA is OK
Found FONA808 (v2) 
Initializing modem...
Waiting for network... OK
Connecting through(apn)---> wholesale----->>> OK and Connected to GPRS
Setting up server----->>  Server Set
Connecting to: iot.wabashnational.com ----->> Fail
Error code number for no connection: -4
```

P.S. I have gone through the existing issue #23 but did not get any solution.
@holgerkoch  and @vshymanskyy Can you help?

tusharvb19, 2017-08-05:
Same issue here!
my serial output:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
Attempting to open comm with ATs
	---> AT
	<--- OK
	---> AT
	<--- OK
	---> AT
	<--- OK
	---> AT
	<--- OK
	---> ATE0
	<--- OK
	---> ATE0
	<--- OK
	---> AT+CVHU=0
	<--- OK
	---> ATI
	<--- SIM800 R13.08

OK

	---> AT+GMM
	<--- SIMCOM_SIM800

OK

FONA is OK
Found FONA808 (v2) 
Initializing modem...
### Unhandled: ³È«(Çb
### Unhandled: ½öÇJõº
### Unhandled: ”ëµgOp
### Unhandled: ´Ãä­ç
### Unhandled: �
'÷‹Ì
### Unhandled: qœ
### Unhandled: ˆò;»
### Unhandled: SIM800 R13.08
Waiting for network...### Unhandled: OK

SIMCOM_SIM8
### Unhandled: 00
### Unhandled: OK
³È
### Unhandled: «(Çb½öÇJõº”ëµgOp´Ãä­ç�
'÷‹Ì
### Unhandled: qœ
ˆò;»
SIMCOM_SIM80
### Unhandled: 0

O
### Unhandled: K

OËK

OK

OK
### Unhandled: SIM800 R13.08
### Unhandled: OK

SIMCOM_SIM800
### Unhandled: OK
³È«(Çb½öÇJõº”ëµgOp´Ã
### Unhandled: ä­ç�
'÷‹Ì	qœ
ˆò;»
### Unhandled: OK
G: 0,1
### Unhandled: OK
 fail

Help needed!

vshymanskyy, 2017-09-02:
I'm not sure if you can use Adafruit_FONA with TinyGSM simultaneously. I think you should try using TinyGSM separately.

closed 2017-09-02 by vshymanskyy

ManOnPajamas, 2018-02-04:
Hi! where you able to fix the problem? Im having a similar issue with my fona808 and arduino uno.