vshymanskyy/TinyGSM #66: [SIM800] Test FAIL on Diagnostics

eWizardII, 2017-08-06:
I'm trying to figure out if the test actually failed or maybe the size of the text file/contents changed since when the original project was started? Because all the responses say OK.

```
OK
AT+CIICR

OK
AT+CIFSR;E0

10.255.78.194

OK
AT+CDNSCFG="8.8.8.8","8.8.4.4"

OK
 OK
Connecting to cdn.rawgit.comAT+CIPSTART=1,"TCP","cdn.rawgit.com",80

OK

1, CONNECT OK
 OK
AT+CIPSEND=1,65

>GET /vshymanskyy/tinygsm/master/extras/test_simple.txt HTTP/1.0
 
DATA ACCEPT:1,65
AT+CIPSEND=1,22

>Host: cdn.rawgit.com
 
DATA ACCEPT:1,22
AT+CIPSEND=1,21

>Connection: close

 
DATA ACCEPT:1,21

+CIPRXGET: 1,1
AT+CIPRXGET=4,1

+CIPRXGET: 4,1,1371

OK
AT+CIPRXGET=2,1,255

+CIPRXGET: 2,1,255,1116
HTTP/1.1 200 OK
Date: Sun, 06 Aug 2017 00:22:02 GMT
Content-Type: text/plain;charset=utf-8
Connection: close
X-Content-Type-Options: nosniff
X-Robots-Tag: none
Access-Control-Allow-Origin: *
ETag: "675136a9bcc9b879e0ebcfbee3eb7bc7745c2e27"
Cache-C
OK
AT+CIPRXGET=2,1,255

+CIPRXGET: 2,1,255,861
ontrol: max-age=86400
Vary: Accept-Encoding
RawGit-Cache-Status: MISS
Server: NetDNA-cache/2.2
X-Cache: MISS

01 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
02 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 012
OK
AT+CIPRXGET=2,1,255

+CIPRXGET: 2,1,255,606
3456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
03 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
04 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstu
OK
AT+CIPRXGET=2,1,255

+CIPRXGET: 2,1,255,351
vwxyz
05 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
06 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
07 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 AB
OK
AT+CIPRXGET=2,1,255

+CIPRXGET: 2,1,255,96
CDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
08 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
09 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
10 #
OK
AT+CIPRXGET=2,1,255

+CIPRXGET: 2,1,96,0
$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz

OK

1, CLOSED
AT+CIPCLOSE=1

ERROR
Server disconnected
AT+CIPSHUT

SHUT OK
GPRS disconnected

************************
 Received: 1371bytes
 Test:     FAIL
************************

```

eWizardII, 2017-08-07:
Two days later revisited my problem - I don't know if this is solved yet - but what I did do was change my baud rates the device SIM800L I made 4800, the Arduino I made 9600. Now I am passing the test.

```
AT
⸮AT
AT

OK
AT&FZE0
AT&FZE0

OK
AT+CPIN?

ERROR
AT+CPIN?

RDY

+CFUN: 1

+CPIN: READY

+CPIN: READY

OK
Connecting to soracom.ioAT+CIPSHUT

Call Ready

SMS Ready

SHUT OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","soracom.io"

OK
AT+SAPBR=3,1,"USER","sora"

OK
AT+SAPBR=3,1,"PWD","sora"

OK
AT+CGDCONT=1,"IP","soracom.io"

OK
AT+CGACT=1,1

OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"10.255.78.194"

OK
AT+CGATT=1

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

OK
AT+CIPRXGET=1

OK
AT+CSTT="soracom.io","sora","sora"

OK
AT+CIICR

OK
AT+CIFSR;E0

10.255.78.194

OK
AT+CDNSCFG="8.8.8.8","8.8.4.4"

OK
 OK
Connecting to cdn.rawgit.comAT+CIPSTART=1,"TCP","cdn.rawgit.com",80

OK

1, CONNECT OK
 OK
AT+CIPSEND=1,65

>GET /vshymanskyy/tinygsm/master/extras/test_simple.txt HTTP/1.0
 
DATA ACCEPT:1,65
AT+CIPSEND=1,22

>Host: cdn.rawgit.com
 
DATA ACCEPT:1,22
AT+CIPSEND=1,21

>Connection: close

 
DATA ACCEPT:1,21

+CIPRXGET: 1,1
AT+CIPRXGET=4,1

+CIPRXGET: 4,1,370

OK
AT+CIPRXGET=2,1,255

+CIPRXGET: 2,1,255,115
HTTP/1.1 200 OK
Date: Mon, 07 Aug 2017 17:47:56 GMT
Content-Type: text/plain;charset=utf-8
Connection: close
X-Content-Type-Options: nosniff
X-Robots-Tag: none
Access-Control-Allow-Origin: *
ETag: "675136a9bcc9b879e0ebcfbee3eb7bc7745c2e27"
Cache-C
OK
AT+CIPRXGET=2,1,255

+CIPRXGET: 2,1,255,860
ontrol: max-age=86400
Vary: Accept-Encoding
RawGit-Cache-Status: MISS
Server: NetDNA-cache/2.2
X-Cache: HIT

01 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
02 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123
OK
AT+CIPRXGET=2,1,255

+CIPRXGET: 2,1,255,605
456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
03 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
04 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuv
OK
AT+CIPRXGET=2,1,255

+CIPRXGET: 2,1,255,350
wxyz
05 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
06 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
07 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABC
OK
AT+CIPRXGET=2,1,255

+CIPRXGET: 2,1,255,95
DEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
08 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
09 #$%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz
10 #$
OK
AT+CIPRXGET=2,1,255

+CIPRXGET: 2,1,95,0
%@`"'~^&-*+=!?.,:;_\|/<>()[]{} 0123456789 ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz

OK

1, CLOSED
AT+CIPCLOSE=1

ERROR
Server disconnected
AT+CIPSHUT

SHUT OK
GPRS disconnected

************************
 Received: 1000bytes
 Test:     PASSED
************************
```

yeffrimic, 2017-08-10:
check the signal 
you have to put another antenna maybe, im using 38400 bauds to "speak" with de sim800l and that work very well on a micro pro board :)

vshymanskyy, 2017-09-02:
Issue resolved, closing.

closed 2017-09-02 by vshymanskyy