vshymanskyy/TinyGSM #68: Not working on ESP32!

tusharvb19, 2017-08-10:
Even the AT debug sketch is not receiving output!
Need help.

vshymanskyy, 2017-09-02:
Could you try with the latest master branch?

vshymanskyy, 2017-09-02:
Which exactly hardware/modem do you use? If the issue persists, please reopen issue and specify all the details. Thank you.

closed 2017-09-02 by vshymanskyy

vshymanskyy, 2017-09-03:
Also, can you confirm that serial on your ESP32 is working at all, i.e, you can issue AT commands?

tusharvb19, 2017-09-05:
I was able to test AT commands using hardware serial port and have successfully tested FONA library.
But when I tried TinyGSM even the AT Debug sketch was not working!

tusharvb19, 2017-09-05:
Will test with the latest branch.

0x1abin, 2017-09-18:
i have same problem, not working in esp32.

vshymanskyy, 2017-09-18:
Just "not working" is not an issue report. @0x1abin , what is you hardware, what sketches did you run an what's the output.

vshymanskyy, 2017-09-18:
@0x1abin sorry I now realized that you created a separate issue.