vshymanskyy/TinyGSM #69: [SIM800] Looping with ERROR AT+CIPRXGET=2,1,63

widoyo, 2017-08-13:
I have successfully HTTP Post to web server every 5 minutes with SIM800L. I put code 
`  client.stop();`
`  modem.gprsDisconnect();`

after doing HTTP Post. Unfortunatelly, sometime I got my system lock up with message:
`ERROR`
`AT+CIPRXGET=2,1,63`
repeated until I reset my CPU.

How best practices if I want to POST data every 5 minutes?
Thank you

vshymanskyy, 2017-09-02:
Please try the latest release of the library, reopen if still have issues.

closed 2017-09-02 by vshymanskyy

JacoBezuidenhout, 2018-06-07:
Good day,

I am using v0.3.3 of the library and I am experiencing the same problem as @widoyo . Any idea why this happens? Any workaround? 

I'm at a point where I want to add a 555 Timer to reset if stuck for too long....

khawajamechatronics, 2018-06-25:
Hi,

I am using v0.3.5 and code froze at 

+CIPRXGET: 2,1,63,365
HTTP/1.1 200 OK
X-Powered-By: Express
Access-Control-Allow-Or
OK
200
AT+CIPRXGET=2,1,63

+CIPRXGET: 2,1,63,302
igin: *
Content-Type: application/json; charset=utf-8
Content
OK
AT+CIPRXGET=2,1,63

+CIPRXGET: 2,1,63,239
-Length: 20
ETag: W/"14-cbmD/auWoSatriSIgZCfCUgg0jw"
set-cook
OK
AT+CIPRXGET=2,1,63

+CIPRXGET: 2,1,63,176
ie: connect.sid=s%3AHppfMVIrDeQ3Z7wuBx

1, CLOSE OK

any one have same issue and any possible solution?

Looking Forward,

mcr-ksh, 2018-08-10:
same here:

```
OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","212.237.59.104","80","CONNECTED"

OK
AT+CIPRXGET=4,1

+CIPRXGET: 4,1,626

OK
AT+CIPRXGET=2,1,63

+CIPRXGET: 2,1,63,563
HTTP/1.1 200 OK
Server: nginx/1.10.3 (Ubuntu
                            ,
1, CLOSED
```

scottjames, 2018-09-01:
I get it too.  The code was working at one point. I'm testing with SIM800L module using Wemos 8266 cpu.  Maybe it's related to a race condition with reading the headers back from the server, or parsing on a character that isn't returned?  I've tried:

  * USB power, and battery power.
  * changing RX buffer from 512 to 1024

The issue seems to consistently stop at the end (or second line?) of reading the server headers.
 
```
... previous lines removed...
OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","198.54.114.134","80","CONNECTED"

OK
AT+CIPRXGET=4,1

+CIPRXGET: 4,1,220

OK
AT+CIPRXGET=2,1,1023

+CIPRXGET: 2,1,220,0
HTTP/1.1 200 OK
Date: Sat, 01 Sep 2018 16r:Egto

1, CLOSED

```