vshymanskyy/TinyGSM #77: [A7] Add support for A7 GPS/GNSS

vshymanskyy, 2017-09-12:
Just created this to track this feature.
PRs are welcome, as usual! ;)

vshymanskyy, 2017-09-17:
I'm gonna to buy a devboard with this module soon, so I'll be able to integrate.

dronecz, 2017-09-23:
Hi, I have A7 dev board and I can receive GPS. I don´t think that support GNSS. Can I anyhow help you with adding GPS support?

vshymanskyy, 2017-09-24:
@dronecz According to https://github.com/vshymanskyy/TinyGSM#features , this is not yet available

vshymanskyy, 2017-09-24:
Probably we just have to enable GPS data stream, and weed it into any GPS decoder library, like: https://github.com/mikalhart/TinyGPSPlus/blob/master/examples/BasicExample/BasicExample.ino

dronecz, 2017-09-24:
I know that is not implemented and I did not saw any references to GNSS for A7 module. There are 2 ways how it can send data stream, one is by dedicated TX pin and second one is by serial communication which is already in use, if you are communicating with module. Anyway, TinyGPS++ is working with A7 module.

mixaz, 2017-11-12:
I use TinyGPS (old one, not ++)  with A7 and TinyGSM - seem to work OK. Yes, using GPS_RX pin

pablods90, 2018-09-03:
As we discussed Gitter.. forked to Forked to https://github.com/pablods90/TinyGSM to start working on this feature.