vshymanskyy/TinyGSM #79: Add support to read SMS

brvdg, 2017-09-12:
I open this, because right now there is no feature to read a SMS.

vshymanskyy, 2017-09-12:
Is there a need to read sms upon reception, or just from the internal memory?

brvdg, 2017-09-12:
Read from internal memory is for me enough. 

I think, to react in the moment when the sms is received is very difficult.  The SimCom module sends an AT command at this moment. If you are in the interaction with the module you get an AT command in the middle of an existing communication.

Take a lock in my code. There is an example to read a sms from the internal storage. For my project is this enough. I check the storage from time to time.

shubhamsharmaa001, 2017-09-20:
I also want that feature.

vshymanskyy, 2017-09-28:
What hardware is under the question?

9H1LO, 2017-10-25:
would be good to have for credit balance checking

vshymanskyy, 2017-10-25:
I use ussd for balance check

9H1LO, 2017-10-25:
i don't think my operator has a ussd code for balance check but i will ask around

do you have a link to a doc/example on ussd query/reply and can this be performed whilst an active gprs & mqtt session is running ?

9H1LO, 2017-10-25:
managed to get ussd code from operator and send with:

String ussd_balance = modem.sendUSSD("*123#");
  Serial.print(ussd_balance);

but get nothing in return

ristomatti, 2017-11-30:
The library looks awesome but receiving and acting on SMS is the sole purpose I bought a SIM800L shield for so I'll join the people wishing for this feature.

I currently use a very old proprietary system to toggle power outlets at a remote location using SMS as the 4G based modem there does not reconnect reliably after a power outage for example. With SMS it's possible to use a prepaid SIM card with no monthly fees for this purpose. 

I believe I can make this work using some other library or AT commands but it's a shame as this library looks the most promising otherwise (and I'm a big fan of Blynk!). Who knows if I'll be able to contribute to this in case I find a good solution for this. :)

PES-TANA, 2017-12-27:
Ajudem sou novo aqui estou a utilizar o módulo A6, como faço ele ler uma mensagem recebida.

vshymanskyy, 2017-12-27:
@PES-TANA Yes, definitely! nostro appetere ut eam. In cum primis corrumpit tincidunt, mel omnes utinam at. Ne qui fabulas laoreet. Et qui dolore epicurei. At pri movet graece utroque, quidam mollis iudicabit eum te. Libris essent euismod in ius, eum ut idque consectetuer.
Maybe I should use google translate at least...

shivsey, 2018-01-13:
Upvote for this feature.  Also @vshymanskyy thank you so much for this library, I'm so thankful to not have to start from scratch.  Seriously thank you!

jashinj, 2018-09-09:
After adding your code in https://github.com/szotsaki/TinyGSM/commit/21845ba49be35058f77c0f54f868cd7238472538 , when I send an SMS to the SIM808 it displays error:

### Unhandled: +CMTI: "SM",14

How do I get the GSMclient to read the latest SMS when this command is received?

It seems the waitResponse function is going to the finalise clause when receiving this command.

szotsaki, 2018-09-15:
@jashinj The library itself just turns on the unsolicited messages when an SMS/MMS arrives; this is what you see (`+CMTI: "SM",14`).

It's the user's responsibility to periodically check their UART line if anything, like an unsolicited message, has been put there. The CMTI message holds the memory index so after you extracted that index you can call `readSmsMessage` with that index.

szotsaki, 2018-10-01:
I created a merged `master` branch with all my pull requests in my [fork](https://github.com/szotsaki/TinyGSM) so if you clone that one, you can use SIM800 SMS handling right away.

Two notes:
* Minimum required Arduino IDE version: 1.8.6.
* C++14 compatibility is required: replace `gnu++11` to `gnu++14` in your `<Arduino IDE directory>/hardware/arduino/avr/platform.txt`.

mayurharge, 2018-11-24:
> Is there a need to read sms upon reception, or just from the internal memory?

> Is there a need to read sms upon reception, or just from the internal memory
Please add THESE FEATURES for sim800
In the ReadSMS feature, these things might be useful
1. Check for number of unread SMS 
2. Read SMS form pointer This function should return the sender (mobile num) and message

mbta009, 2019-02-04:
@szotsaki 
Hello and respect
I tested your code. When compiling, I encountered the following. It should be noted that I am using the visual studio for arduino coding (visual micro), however, I did the changes you have made to convert gnu 11 to gnu 14. I also use the IDE 1.8.7 version. The eyeball is as follows:

Error error: invalid use of 'auto
error: invalid conversion from 'const __FlashStringHelper *' to 'int' [-fpermissive]

mayurharge, 2019-02-05:
can you please send your actual code

Thanks & Regards,
   Mayur Harge <http://www.linkedin.com/in/mayurharge>
+919096028050
+918668423824

mbta009, 2019-02-05:
@mayurharge  
just Download and install library [https://github.com/szotsaki/TinyGSM](https://github.com/szotsaki/TinyGSM)
and compile any example on this Library (Example on szotaski Repository equal to vshymanskyy/TinyGSM ) error in mentioned in last post for Library szotsaki/TinyGSM 
but in vshymanskyy/TinyGSM no have error 
Thanks.

mbta009, 2019-02-05:
i want use MQTT and read/delete SMS feature in my sim800 project. Tomorrow I must deliver my project. If you have an existing source code or need to make changes to the library, share it here. thank you

mayurharge, 2019-02-05:
@mbta009 it seems that data type of defined function and the return value is not same.
so you might need to edit the data type for each function

mayurharge, 2019-02-05:
if you just want to use sms feel free to use my forked library

mbta009, 2019-02-05:
@mayurharge  Thanks. Check Example code of your lib, that's Compile without error. 
my main code work with vshymanskyy/TinyGSM but with your lib have error : 
ago mentioned:
> @mbta009 it seems that data type of defined function and the return value is not same.
> so you might need to edit the data type for each function
Can you explain more?
Thanks. 
```
Severity	Code	Description	Project	File	Line	Suppression State
Error (active)	E1696	cannot open source file "stddef.h"	Logger_Sim800	c:\Users\MBTA\Documents\Arduino\libraries\ArduinoJson-master\src\ArduinoJson\JsonVariant.hpp	7	
Error		465:38: error: 'REG_OK_ROAMING' was not declared in this scope		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmClientSIM800.h	465	
Error		465:18: error: 'REG_OK_HOME' was not declared in this scope		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmClientSIM800.h	465	
Error		462:36: error: 'REG_OK_TCP' was not declared in this scope		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmClientESP8266.h	462	
Error		462:18: error: 'REG_OK_IP' was not declared in this scope		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmClientESP8266.h	462	
Error		428:38: error: 'REG_OK_ROAMING' was not declared in this scope		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmClientBG96.h	428	
Error		428:18: error: 'REG_OK_HOME' was not declared in this scope		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmClientBG96.h	428	
Error		427:38: error: 'REG_OK_ROAMING' was not declared in this scope		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmClientUBLOX.h	427	
Error		427:18: error: 'REG_OK_HOME' was not declared in this scope		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmClientUBLOX.h	427	
Error		404:38: error: 'REG_OK_ROAMING' was not declared in this scope		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmClientM590.h	404	
Error		404:18: error: 'REG_OK_HOME' was not declared in this scope		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmClientM590.h	404	
Error		395:38: error: 'REG_OK_ROAMING' was not declared in this scope		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmClientA6.h	395	
Error		395:18: error: 'REG_OK_HOME' was not declared in this scope		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmClientA6.h	395	
Error		38:28: error: redefinition of 'const char GSM_OK []		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmCommon.h	38	
Error		38:28: error: redefinition of 'const char GSM_OK []		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmCommon.h	38	
Error		38:28: error: redefinition of 'const char GSM_OK []		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmCommon.h	38	
Error		38:28: error: redefinition of 'const char GSM_ERROR []		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmCommon.h	38	
Error		38:28: error: redefinition of 'const char GSM_ERROR []		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmCommon.h	38	
Error		355:36: error: 'REG_OK_TCP' was not declared in this scope		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmClientESP8266.h	355	
Error		355:18: error: 'REG_OK_IP' was not declared in this scope		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmClientESP8266.h	355	
Error		33:6: error: multiple definition of 'enum RegStatus		C:\Users\MBTA\Documents\Arduino\libraries\TinyGSM-master\src\TinyGsmClientBG96.h	33	

```

mbta009, 2019-02-05:
@ristomatti Thanks for test. yes my compiler also error at "auto", your compile, Did not get the error? 
no problem for remove "auto" in performance Library?

ristomatti, 2019-02-05:
@mbta009 To be honest, I'm not an expert on C++ (quite the contrary). But my understanding is that adding a return type to a lambda function (`-> <type>`) can be used for extra type safety. Setting a return type `auto` does not seem to make sense to me as it by definition means that the compiler should be able to figure out the correct type based on the code. That's the point of having the variable type of `const auto`.

ristomatti, 2019-02-05:
@mbta009 If you were worried about the effect on performance, it should not make any difference. The return type is just for compile time type safety (static error checking of the code).

ristomatti, 2019-02-05:
Another way to make it compile is to change the line to:

```cpp
const auto convertMstToString = [](const MessageStorageType &type) -> const __FlashStringHelper* {
```

But that just looks ugly.

mbta009, 2019-02-05:
@ristomatti  
Thank you for your information. I'm currently using the https://github.com/mayurharge/TinyGSM library. But now I have time limits for the end of the project. After completing the project, I will test https://github.com/szotsaki/TinyGSM, and I will quote the result of the removal of "auto". Thanks

VinceKezel, 2019-10-15:
+1 for read new SMS... from memory is fine, don't necessarily need to trigger an event right at time of SMS reception.

emigbur, 2019-10-15:
Hi
I used  to trigger an event right at time of SMS reception for immediate action like open a door 
`while (SerialAT.available() > 0) {
    int index = modem.newMessageIndex(0);
    if (index > 0) {
      String SMS = modem.readSMS(index);
      String ID = modem.getSenderID(index);
      DBG("new message arrived from :");
      DBG(ID); DBG("Says"); DBG(SMS);
      SerialMon.println(sizeof(SMS.toInt()));
`
And work fine, just not sure if olds SMS are deleted . the library deleted itself?

mmz5757, 2019-10-19:
@vshymanskyy @szotsaki, I am trying to read SMS with active TinyGSM MQTT connection but I couldn't do it. Can you guys provide a solution?

alexgavs, 2020-01-11:
> @mbta009 It seems I was a few minutes too late. I've been planning to test @szotsaki 's fork also, so this discussion got me interested to see if it'll build ok on my system. I am using PlatformIO for development and got the same error on compilation (MQTTClient.ino example).
> 
> The only thing I had to do to get it to compile was remove the (AFAIK optional) return type declaration `-> auto` on line 954 of TinyGSMClientSim800.h. So change from:
> 
> ```c++
> const auto convertMstToString = [](const MessageStorageType &type) -> auto {
> ```
> 
> To:
> 
> ```c++
> const auto convertMstToString = [](const MessageStorageType &type) {
> ```
> 
> I don't have the hardware at hand to test it though but at least now it compiles to both ATmega2580 and ESP8266. I did not even need to modify anything in `platform.txt`.

put to platformio.ini
src_build_flags = -Os -std=gnu++14

after 
`const auto convertMstToString = [](const MessageStorageType &type) -> auto {`
 is approved

ristomatti, 2020-01-11:
Thanks for the tip @alexgavs !

szotsaki, 2020-01-12:
Hello guys,

Sorry for this taking so long; I've updated my fork and besides merged in the latest version of this variant, I've removed the `-> auto` return value indicator so no C++14 is required from now on.

Please, test it at https://github.com/szotsaki/TinyGSM and tell me if you bumped into an issue and I'll try to fix it.

gshakourian, 2020-04-27:
Hello

I use the library below:
https://github.com/szotsaki/TinyGSM

But I have trouble using the message reading function.
Can you show the reading message by an example?
Please help me.

thanks

garudaonekh, 2021-11-17:
Hello, this look like the newest update for reading SMS but can't get it to work properly. Any good example is much appreciated.

https://github.com/pnb990/TinyGSM