vshymanskyy/TinyGSM #81: [SIM800]ESP32 not work.

0x1abin, 2017-09-18:
i use it in hardware Serial2.

`HardwareSerial Serial2(2);
#define SerialAT Serial2`

The example AT_Debug isn't wrok return.
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:708
load:0x40078000,len:0
load:0x40078000,len:11460
entry 0x400789f4
Trying baud rate 115200 ...
Modem responded at rate 115200
***********************************************************
 You can now send AT commands
 Enter "AT" (without quotes), and you should see "OK"
 If it doesn't work, select "Both NL & CR" in Serial Monitor
***********************************************************
-> AT
-< ⸮:

0x1abin, 2017-09-18:
and the blynk example isn't compiling.

vshymanskyy, 2017-09-18:
For the Blynk library - please post an issue on Blynk library github page.

vshymanskyy, 2017-09-18:
@0x1abin is this issue specific for TinyGSM library? Are you able to send AT commands manually (or using any other example sketch)? If yes, please provide a working sketch.

vshymanskyy, 2017-09-19:
@0x1abin I need your help to understand the issue.

vshymanskyy, 2017-09-24:
Closed due to lack of interest

closed 2017-09-24 by vshymanskyy

thomasklingbeil, 2017-10-10:
I can confirm the exact same issue.
ESP32 with N590 connected to HardwareSerial on pins 32/33.

When using a simple sketch, which only sends data back and forth between the default serial console (USB) and the HardwareSerial, everything works properly.

With the TinyGSM library, a lot of random characters show up during the communication, rendering all results unusable.