vshymanskyy/TinyGSM #82: [SIM800/SIM808] GPRS disconnect not complete

brvdg, 2017-09-18:
Hi,

if I connect vie GPRS and want to disconnect, the GPRS connection is not complete closed. The module is blinking like connected.
The problem is, TinyGSM doesn't detach the GPRS service.

Solution:
Please add 

`sendAT(GF("+CGATT=0"));`

to the disconnect code.

Thanks

vshymanskyy, 2017-09-19:
Thank you! Will do.
BTW, do you know how to check GPRS status reliably on SIM8xx?

brvdg, 2017-09-19:
Which status do you exactly mean?
The GPRS network status is the same like the network status (AT+CGREG? instead of AT+CREG?).

AT+CGATT? replies the GPRS status. 0 = detached, 1 = attached. 
That is what the LED indicates on the modules.

GeorgianBorca, 2017-09-20:
This is how I check if the my modem is connected to internet or not.

`boolean TrackME_GSM::isOnline() {
	byte conn= 0;
	if (!sendAT(F("AT+SAPBR=2,1"), F("OK"), 3000)) { //+SAPBR: 1,3,"0.0.0.0"
		return false;
	}	
	if (strstr_P(gsmBuffer, PSTR ("BR:"))) {
		strtok_P(gsmBuffer, PSTR (" "));
		strtok_P(NULL, PSTR (","));
		conn = atoi(strtok_P(NULL, PSTR (",")));
	}
	if (conn == 2 || conn == 3) {
		return false;
	} else if (conn == 0 || conn == 1) {
		return true;
	} else {
		return false;
	}
	
	return false;
}`

vshymanskyy, 2017-09-24:
Please check it on the latest master branch

commit 0389264e (2017-09-24): Add isNetworkConnected(), isGprsConnected(). Fix #82
commit df2b5cae (2017-09-24): Add isNetworkConnected(), isGprsConnected(). Fix #82
closed 2017-09-24 by vshymanskyy