vshymanskyy/TinyGSM #86: Get rid of the Arduino String class

karioja, 2017-09-23:
I'm doing some minor testing whether TinyGSM works with the FreeRTOS Arduino port and running into problems. The data object in waitResponse (an Arduino String) is always empty.

It seems to related to
https://stackoverflow.com/questions/39551941/arduino-string-not-working-inside-freertos-task
and/or
https://github.com/feilipu/Arduino_FreeRTOS_Library/issues/3

Getting rid of the String class would make TinyGSM a bit more portable.

vshymanskyy, 2017-09-24:
@karioja - it would be great to get rid of String, especially if it makes code more efficient. It will impact all supported modules. I'm ready to review PRs for this.

karioja, 2017-09-24:
Initial testing seems to suggest that replacing String with statically allocated char buffers seems to work. I've re-implemented parts of the waitResponse using a static response buffer and couple of the modem interface methods to return a char* to a static buffer holding the value for modem info, imei etc. (all functions which return String).

anatom74, 2018-03-18:
@karioja Any progress running TinyGSm+FreeRTOS? If possible, I would like to test your fork...

mixaz, 2018-03-18:
I'm using TinyGSM on FreeRTOS, I do not remember I made any patches for it. 
But I would recommend using other GSM library for FreeRTOS, it looks more professional: http://majerle.eu/projects/gsm-at-commands-parser-for-embedded-systems. I'll plan to use it in my next GSM project, just I found it when it was too late to switch to a new library.

karioja, 2018-03-26:
I've stopped working on this due to lack of time. I'm still hoping there's a way to run TinyGSM with the Arduino FreeRTOS using some workaround other than completely rewriting TinyGSM. @mixaz which specific FreeRTOS version did you use?

mixaz, 2018-03-26:
FReeRTOS 9.0.0, port for STM32

anatom74, 2019-08-12:
seems that fork from [kramarb ](https://github.com/kramarb/TinyGSM)is working (I tested it with SIM868), so it might be a good start

BracketJohn, 2019-12-23:
See my write up on https://github.com/vshymanskyy/TinyGSM/issues/352 for other problems that occur from the usage of `String`. Also see https://github.com/vshymanskyy/TinyGSM/issues/255 for a problem arising from the faulty String casting implemented by Arduino.

SRGDamia1, 2020-04-15:
Version 0.10.1 dramatically reduces the use of **S**trings, but they haven't been completely eliminated.