vshymanskyy/TinyGSM #87: MySQL Connector & your lib

dronecz, 2017-09-23:
Hi, I just find your great library and I would like to use it with MySQL_Connector_Arduino library (https://github.com/ChuckBell/MySQL_Connector_Arduino) but I can not figure it out if I can use GPRS to send data to db. I think it should be possible but I´m not sure. I can send data to db by wifi, but I would like to use GPRS for remote location of my device. 

Can you please let me know if yes or no? 

Thanks

vshymanskyy, 2017-09-24:
Did you try taking some basic example like https://github.com/ChuckBell/MySQL_Connector_Arduino/blob/master/examples/connect/connect.ino , and change it to use TinyGSM? Did it work? What hardware do you use? What errors do you get?

vshymanskyy, 2017-09-24:
Please continue discussion on this topic in our chat: https://gitter.im/tinygsm/Lobby

closed 2017-09-24 by vshymanskyy

dronecz, 2017-09-24:
Yes, I did and Arduino keep restarting. I´m using Arduino Uno & SIM900 Shield. I got no error, Arduino is just restarting.