vshymanskyy/TinyGSM #91: [M590] LIbrary not working on ESP32

thomasklingbeil, 2017-10-10:
As already described in issue #81, using HardwareSerial on ESP32 does not work with the library.
If transferring the data "manually" in a simple sketch, everything works fine.

With the TinyGSM library, a lot of random characters show up during the communication, rendering all results unusable.

thomasklingbeil, 2017-10-11:
The problem seems to be related to the implementation of HardwareSerial for ESP32. With the most recent git state, things seems to be better (see also https://github.com/espressif/arduino-esp32/issues/665)

vshymanskyy, 2017-10-11:
I'll re-check this once I get some spare time. Thanks!

papawattu, 2018-01-06:
Hi 

I am getting the same issue, is there any update on this?

[image]

papawattu, 2018-01-08:
Looks like my issue was unrelated to this maybe, I am using a Adafruit Huzzah 32.  I switched to using UART1 (2nd one) and identified the pins

```
HardwareSerial Serial1(1);

#define SerialAT Serial1

TinyGsm modem(SerialAT);
TinyGsmClient client(modem);

void setup() {
  SerialAT.begin(115200,SERIAL_8N1,16,17,false);
  delay(3000);
  modem.restart();
}

```

This now works fine.  Hope this helps someone else.

Jamie.

closed 2018-02-03 by vshymanskyy

ankgt, 2018-03-10:
@papawattu I'm facing a similar issue, and trying to understand your solution. If you switched to UART1, does that mean you can no longer use the USB port on your board to monitor the program? Thanks!

DjordjeMandic, 2018-03-23:
@papawattu thats not needed, look at source code of hardwareserail.h `void begin(unsigned long baud, uint32_t config=SERIAL_8N1, int8_t rxPin=-1, int8_t txPin=-1, bool invert=false);` im using esp32 and all AT commands work from debug but when i wanna use library it all gets fked up

abhaysbharadwaj, 2018-12-01:
@papawattu , @vshymanskyy , I can confirm that TinyGSM is working flawlessly on my ESP32.
As per comment, I have added the following before setup():
`HardwareSerial` gsmSerial(1); // (1) implies serial 1
`TinyGsm` modem(gsmSerial);
`TinyGsmClient` client(modem);
`PubSubClient` mqtt(client);

and in setup:
`gsmSerial.begin(115200,` SERIAL_8N1, 18, 19, false); // i am remapping serial1 pins to 18 and 19

[image]

jrutke, 2018-12-09:
I have a problem with gprs connection in my ESP32. Don't find the modem. WIth AT command's its working. Can someone help me?

@abhaysbharadwaj, how did you use?

vitzaoral, 2018-12-09:
I can't connect _SIM800L EVB_ version to my Espressif _ESP32-WROOM-32_ developer kit. I'm using _UART 2_ connection (SIM800L TX -> ESP32 TX2, SIM800L RX -> ESP32 RX2, [picture here](https://res.cloudinary.com/vitaz/image/upload/v1544393116/IMG_9415.jpg)). I think there is no problem with power supply, I can call to SIM800L, it blinks every 3sec so it's connected to network.

I'm using code as was described above:
```
HardwareSerial gsmSerial(2);
TinyGsm modem(gsmSerial);

void setup()
{
 // Set GSM module baud rate
  gsmSerial.begin(115200, SERIAL_8N1, 16, 17, false);
....
```

There is no response from SIM800L. Am I doing something wrong, or is _EVB_ version somehow different from "classic" red SIM800l?
Thank you.

remenyo, 2019-04-20:
@vitzaoral 
You connected RX to RX and TX to TX, but you have to connect RX into TX, and TX into RX.
[Wiring and Hardware of UART devices](https://learn.sparkfun.com/tutorials/serial-communication#wiring-and-hardware)

sass00, 2022-01-23:
Hey guys, 

Have a problem with my ESP32. Using DevKitC V4 with Wrower. Trying to hook up SIM800L. My unit is connected, i can type AT commands through USB-TTL. Light flashes as its connected to the network. 

My device is defined as serial2 on pins 16, 17. When i start my ESP it cant initialize modem. Tried example sketeches of both TinyGSM and FONA library. Works normally directly through serial on arduino (i short RST to GND) so i know the SIM module is working. 

#define SerialMon Serial
#define TINY_GSM_MODEM_SIM800
#include <TinyGsmClient.h> // 
HardwareSerial SerialGSM(2);
TinyGsm modemGSM(SerialGSM);
const int BAUD_RATE = 9600;
long int millisRefCon, millisUserResp;
bool flagCon = false, flagUserResp = false;
const int relayPin1 = 14, relayPin2 = 12, relayPin3 = 13, relayPin4 = 27, RX_PIN = 16, TX_PIN = 17;
const char *APN = "internet";
const char *USER = "mobitel";

This is my code. Tried different baud rates, inverting RX/TX pins, different UART, ... nothing works. I'm just using the sim800 to trigger relays and send temp and humidity status. 

I am powering my SIM800L from a outer power supply with 3A max. My wiring is GND->GND; TX->17; RX->16; GND->GND. Tried also rreversing RX/TX. I'm using V2 of the SIM800 (EVB version). 

I'm a bit depressed allready :)

CcKefa, 2024-01-09:
Was this issue resolved? I'm having the same issue and using UART1 does not solve it.

live-alchemy, 2024-02-23:
@CcKefa @sass00 took me some time to get this working, I'll paste the important parts for reference maybe it helps you as well. I had to instantiate my serial connection to the LTE module as a `HardwareSerial` and pass it `(1)` so it would use the Uart1 option. Then later, in the `SerialAt.begin()` call, I pass the RX & TX pins that map to Uart1 on [my device](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/_images/ESP32-S3_DevKitC-1_pinlayout_v1.1.jpg), in this case 18 and 17. Notice in the diagram how these are labelled U1RXD and U1TXD. Maybe that's obvious but for me it was a missing link, the diagrams [here](https://microcontrollerslab.com/esp32-uart-communication-pins-example/) helped me as well.

Also, my LTE module uses 115200 baud rate, so I have passed that in as well, but if you are unclear what baud rate your LTE module uses then use [this AT_Debug sketch](https://github.com/vshymanskyy/TinyGSM/blob/master/tools/AT_Debug/AT_Debug.ino) to try and find that first

```
#define SerialMon Serial
HardwareSerial SerialAT(1);

#define PIN_TX              17
#define PIN_RX              18

#include <TinyGsmClient.h>

TinyGsm        modem(SerialAT);
TinyGsmClient  client(modem);

void setup() {
    SerialMon.begin(115200);
    SerialAT.begin(115200, SERIAL_8N1, PIN_RX, PIN_TX, false);
}
```

lukwagohakim, 2024-07-22:
has anyone used a7672e, is there an issue with its connection to server for both webClient and mqttClient examples? below is my mqttClient code
`/**************************************************************
 *
 * For this example, you need to install PubSubClient library:
 *   https://github.com/knolleary/pubsubclient
 *   or from http://librarymanager/all#PubSubClient
 *
 * TinyGSM Getting Started guide:
 *   https://tiny.cc/tinygsm-readme
 *
 * For more MQTT examples, see PubSubClient library
 *
 **************************************************************
 * This example connects to HiveMQ's showcase broker.
 *
 * You can quickly test sending and receiving messages from the HiveMQ webclient
 * available at http://www.hivemq.com/demos/websocket-client/.
 *
 * Subscribe to the topic GsmClientTest/ledStatus
 * Publish "toggle" to the topic GsmClientTest/led and the LED on your board
 * should toggle and you should see a new message published to
 * GsmClientTest/ledStatus with the newest LED status.
 *
 **************************************************************/

// Select your modem:
// define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM868
// #define TINY_GSM_MODEM_SIM900
// #define TINY_GSM_MODEM_SIM7000
// #define TINY_GSM_MODEM_SIM7000SSL
// #define TINY_GSM_MODEM_SIM7080
// #define TINY_GSM_MODEM_SIM5360
// #define TINY_GSM_MODEM_SIM7600
#define TINY_GSM_MODEM_A7672X
// #define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_SARAR4
// #define TINY_GSM_MODEM_SARAR5
// #define TINY_GSM_MODEM_M95
// #define TINY_GSM_MODEM_BG95
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_MC60
// #define TINY_GSM_MODEM_MC60E
// #define TINY_GSM_MODEM_ESP8266
// #define TINY_GSM_MODEM_ESP32
// #define TINY_GSM_MODEM_XBEE
// #define TINY_GSM_MODEM_SEQUANS_MONARCH

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

#include <HardwareSerial.h>

HardwareSerial SerialAT(2);

// Define the DTR and RI pins used for hardware flow control
const int dtrPin = 18;
const int riPin = 19;

// See all AT commands, if wanted
// #define DUMP_AT_COMMANDS

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon

// Range to attempt to autobaud
[... 247 lines cut ...]
Model: A7672E-FASE
Revision: A7672M7_V1.11.1
IMEI: 8622050597032~
Modem Info: 
Waiting for network...[23689] ### Unhandled: SMS DONE
[24939] ### Unhandled: <@`x
+Xpx`AT+CR
[26189] ### Unhandled: AT+CREG?

ERROR
[27444] ### Unhandled: ~
 success
Network connected
Connecting to internet[88498] ### Unhandled: AT+CGACT=1,1

OK

PB DONE
 success
GPRS connected
=== MQTT NOT CONNECTED ===
Connecting to broker.hivemq.com[92567] ### Unhandled: AT+CIPACK=0

+IP ERRO
 fail

commit ba859869 (2018-02-03): Update Readme, Fix #91