vshymanskyy/TinyGSM #92: A6 MQTT Problems

fuyuhin, 2017-10-11:
1. The library works when the firmwork of the A6 is V03.03.20161229019H03. The problem is that the payload of MQTT has a limit of bytes. Mine is 11 bytes. More than 11 bytes, the MQTT callback has NO reaction.
2. With the new firmwork of A6, V03.05.20170090814H38, TinyGSM can not connect to the broker.

vshymanskyy, 2017-10-11:
How do you change the fw? Can you send ne the files?

fuyuhin, 2017-10-12:
You can download it here: http://wiki.ai-thinker.com/gprs_download
By the way, the tool only works on Windows with USB to serial chip CH340.
With other USB to serial chip, the tool fails.

cimba007, 2018-09-22:
Just wanted to confirm that mqtt connect is not working with A6 DEV-Firmware (Modem: Ai Thinker Co.LTD A6  V03.06.20171127R)