vshymanskyy/TinyGSM #93: [M590] MQTT on ESP8266 crashes on connection problems

thomasklingbeil, 2017-10-12:
It seems that on connection problems, when using MQTT, the ESP crashes.
Receiving/Sending MQTT messages works properly at first, but as soon as the module signals **"+TCPCLOSE:1,Link Closed"** the module crashes, maybe a problem when reconnecting? After the automatic reboot everything goes back to normal.

```
...
+TCPRECV:1,22,0 GsmClientTest/led1
Message arrived [GsmClientTest/led]: 1
AT+TCPSEND=1,28

>0 GsmClientTest/ledStatus0

OK

+TCPSEND:1,28

+TCPRECV:1,2,� 

+TCPRECV:1,22,0 GsmClientTest/led0
Message arrived [GsmClientTest/led]: 0
AT+TCPSEND=1,28

>0 GsmClientTest/ledStatus1

OK

+TCPSEND:1,28

+TCPRECV:1,22,0 GsmClientTest/led1
Message arrived [GsmClientTest/led]: 1
AT+TCPSEND=1,28

>0 GsmClientTest/ledStatus0
AT+TCPSEND=1,2

ERROR

OK

+TCPSEND:1,28

+TCPRECV:1,22,0 GsmClientTest/led0
Message arrived [GsmClientTest/led]: 0
AT+TCPSEND=1,28

>0 GsmClientTest/ledStatus1

OK

+TCPSEND:1,28

+TCPCLOSE:1,Link Closed
AT+TCPCLOSE=1

+TCPCLOSE:Error

Exception (29):
epc1=0x40202186 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000015 depc=0x00000000

ctx: cont 
sp: 3ffef860 end: 3ffefb10 offset: 01a0

>>>stack>>>
3ffefa00:  00000000 3ffee8b4 3ffefa70 4020217c  
3ffefa10:  00000000 00000000 00000000 40203f27  
3ffefa20:  3ffe8671 3ffe8669 0005d4e2 000003e8  
3ffefa30:  3ffe8671 00000000 3ffefa70 40203fae  
3ffefa40:  3ffe8531 00000000 00000000 000003e8  
3ffefa50:  3ffe8671 3ffe8669 00000000 4020223c  
3ffefa60:  00000000 00000000 3ffe8530 4020479c  
3ffefa70:  3fff0e74 0000004f 0000000c 3ffeeae4  
3ffefa80:  3ffee8b4 00000000 3ffeeaf0 3ffe8514  
3ffefa90:  00000001 3ffeeae4 402042bc 3ffeeae4  
3ffefaa0:  3fffdad0 00000000 3ffee794 40202298  
3ffefab0:  00000000 3ffe8514 3ffee8b4 402025a5  
3ffefac0:  3fffdad0 00000000 3ffee794 402025ed  
3ffefad0:  40201eec 3ffeea28 3ffee6d8 4020473d  
3ffefae0:  00000000 00000000 3ffee6d8 4020252b  
3ffefaf0:  3fffdad0 00000000 3ffeeadc 40204308  
3ffefb00:  feefeffe feefeffe 3ffeeaf0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
�Initializing modem...
ATE0

OK
AT+CFUN=15

SOCKETS: IPR STOPPED

SOCKETS: IPR STOPPED

OK

SOCKETS: IPR STOPPED

��#
+�#
+XDRVI: 9,1,7
����#
+XDRVI: 9,1,7
�
MODEM:STARTUP

+PBREADY
ATE0
ATE0

OK
AT&FZE0

OK
AT+CPIN?

+CPIN: READY

OK
Waiting for network...AT+CREG?
...

```

lumtech, 2017-10-22:
i have the same problem only when i use example WebClient on ESP8266, but not always. 
Sometimes the ESP8266 gets a string with data - about once in ten, sometimes crashes on connection , but basically gets an empty string. I tried other ESP8266, changed the softwareserial pins on ESP and libs,  but again there is unstable work. The same result if instead of M590 use A6 module. With arduino nano works fine.

vshymanskyy, 2017-10-28:
PRs are welcome for this...

vanbwodonk, 2018-03-23:
I think that common issue on esp8266. RTOS need to handle some commands every xxx microseconds. So add delay in infinite loop maybe help.