vshymanskyy/TinyGSM #94: SIM800L Readwebpage (GPRS) missing first characters

jovandob, 2017-10-18:
Hi,

board Arduino UNO 5V/16 MHz and SIM800L module, LiPo powered.

All works, but when connecting to cdn.rawgit.com/vshymanskyy/tinygsm/master/extras/logo.txt
there are missing first few characters in first row of printed response, like this:

	---> AT+HTTPREAD
	<--- +HTTPREAD: 121
__            _____  _____  _____
    |  | |\ | \_/ |  ___ |_____ |  |  |
    |  | | \|  |  |_____| _____||  |  |

****
	---> AT+HTTPTERM
	<--- OK

This is the besct case with speeds Serial.begin(115200); and fonaSerial->begin(9600);

Any ideas?

jovandob, 2017-10-18:
Not copied well, here is the problem more visible. Site returns 123456789.
Notice also errors in AT response:

FONA> w
NOTE: in beta! Use small webpages to read!
URL to read (e.g. www.adafruit.com/testwifi/index.html):
http://mysite.com/sensor/write.php?value=123456789
****
	---> AT+HTTPTERM
	<--- ERROR
	---> AT+HTTPINIT
	<--- OK
	---> AT+HTTPPARA="CID"
	<--- OK
	---> AT+HTTPPARA="UA"
	<--- OK
	---> AT+HTTPPARA="URL"
	<--- OK
	---> AT+HTTPACTION=0
	<--- OK
Status: 200
Len: 9
	---> AT+HTTPREAD
	<--- +HTTPREAD3 9
6789
****
	---> AT+HTTPTERM
	<--- OK
FONA>

vshymanskyy, 2017-10-18:
`AT+HTTPREAD` has nothing to do with TinyGSM library. We do not use this AT command at all.

closed 2017-10-18 by vshymanskyy

jovandob, 2017-10-19:
Uuups. Sorry about that, I experimented at the same time with another library and mistaken them.
Your library works perfect here. Great job.