vshymanskyy/TinyGSM #95: TinyGSM + telegram

EgorChadov, 2017-10-21:
Hi! Anybody tried use sim800l + tinygsm with telegram to create bot? I look at HttpsClient example and changed it to send GET request to api.telegram.org. And server answer is "400 Bad Request - The plain HTTP request was sent to HTTPS port Server".

Maybe anybody who uses TineGSM with telegram can show example?

vshymanskyy, 2018-04-23:
sim800l supports SSL for quite some time already, you should be able to do what you wanted ;)

SRGDamia1, 2020-04-15:
I'm closing this old issue.  If you're still having trouble, reopen it.

closed 2020-04-15 by SRGDamia1

rayellam, 2023-05-26:
Im also having trouble getting SIM800l to work with telegram, im throwing a guess that SIM800l only supports TLS1.0? Any ideas anyone?